arduino/Arduino #7025: Carpeta DMCcache, ayuda por favor! lo agradeceria mucho mucho

bocha777, 2017-12-14:
Descubri que el error que tengo de no reconocerme la placa a pesar de tenerla conectada y reconocida, debo copiar y pegar los archivos de la carpeta sdk_fix debo reemplazarlos en la carpeta DMCache, que no la tengo de todas maneras la estoy buscando por todas partes, por el momento no la estoy encontrando. por eso agradeceria mucho si hay alguien que haya tenido el mismo inconveniente y ya tenga la misma carpeta DMCache, que tengan buenos dias y muchas gracias! muyy bueno el programa excelente solo me falta corregir este error

facchinm, 2017-12-14:
Hi @bocha777 , it doesn't look like your problem is Arduino related. If it is, please repost it in the Spanish section of the forum http://forum.arduino.cc/index.php?board=32.0 . 
Thanks

closed 2017-12-14 by facchinm

bocha777, 2017-12-14:
Hola , que quiere decir con que no esta relacionado con el Arduino? https://www.youtube.com/watch?v=ariJHH1gMg4&t=479s como podra ver en el video del link en el minuto  dice de que el Arduino tiene unos fallos que hay que corregirlos para que entienda bien la placa, y veis la carpeta sdk:fix copiar los archivos y pegarlos en la carpeta luego de apretar inicio+r %appdata% buscando asi la carpeta DMCache que me falta para copiar y reemplazar los archivos que contiene sdk_fix para volver a compilarlo y quitar el error, desde ya te agradezco mucho la contestacion. me seria de mucha ayuda que me comentes donde encontrar la libreria  . te agradezco mucho loco! gracias por compartir tus conocimientos conmigo y con la gente que pueda llegar a tener el mismo inconveniente