arduino/Arduino #7026: println does not send carriage return

marcspeth, 2017-12-14:
Yesterday I updated from version 1.6.5 to version 1.8.5 and my project that uses an Arduino Leonardo to send text via keyboard output no longer sends carriage returns at the end of a PRINTLN command.  (yes, using version 1.8.5 I added #include keyboard.h  and  #include mouse.h)

I also tried expressly sending a Carriage Return with Keyboard.print(13); and that does not send a Carriage Return either.

I then took the project back to version 1.6.5 (and commented out the #include keyboard.h  and  #include mouse.h ) and it compiles and works correctly once again.

per1234, 2017-12-14:
Seems to be related to https://github.com/arduino/Arduino/issues/6922.

facchinm, 2017-12-15:
Could you move the issue to https://github.com/arduino-libraries/keyboard? Thanks!

closed 2018-02-28 by per1234