arduino/Arduino #7030: preproc.substitute_unicode

serg512, 2017-12-16:
Hi, dear!
Setting "preproc.substitute_unicode=false" works properly only if sketch is not saved.
If sketch is saved, some functions, such as "lcd.print(string)", still send a unicode (two bytes char) to display.
If sketch is not saved (any change, like one more space in comment string) - it's all right.
PS. I'm using Cyrillic codepage 1251

per1234, 2017-12-16:
Maybe related to https://github.com/arduino/Arduino/issues/4231

Bodmer, 2018-01-02:
My tests on IDE 1.8.5 indicate that preproc.substitute_unicode=true or false has no effect.

Edit, do not save a sketch, compile/upload: then UTF-8 encoding **is not** used regardless of setting
Save the sketch, compile/upload and UTF-8 encoding **is used** regardless of the setting

serg512, 2018-01-03:
Maybe matter:
1. Cyrillic codepage (1251)
2. LiquidCrystal library
3. Localized operating system (windows 7-10 with Russian locale)

facchinm, 2018-01-08:
Hi @serg512 and @Bodmer , since the issue is almost a duplicate of #4231, could you move there your information (which are useful to reproduce/debug the issue) and close this one? Thanks