arduino/Arduino #7031: that it!!! at start in the terminal

slava8686, 2017-12-17:
/usr/bin/build-classpath: error: Could not find apache-commons-lang3 Java extension for this JVM
/usr/bin/build-classpath: error: Could not find bcpg Java extension for this JVM
/usr/bin/build-classpath: error: Could not find bcprov Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jackson-core Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jackson-databind Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jackson-modules-base Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jmdns Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jsch Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jsemver Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jssc Java extension for this JVM
/usr/bin/build-classpath: error: Could not find guava Java extension for this JVM
/usr/bin/build-classpath: error: Could not find objectweb-asm Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jackson-annotations Java extension for this JVM
/usr/bin/build-classpath: error: Could not find jakarta-commons-httpclient Java extension for this JVM
/usr/bin/build-classpath: error: Could not find rsyntaxtextarea Java extension for this JVM
/usr/bin/build-classpath: error: Could not find batik Java extension for this JVM
/usr/bin/build-classpath: error: Could not find xmlgraphics-commons Java extension for this JVM
/usr/bin/build-classpath: error: Some specified jars were not found
java.lang.NoClassDefFoundError: com/fasterxml/jackson/core/JsonProcessingException
        at processing.app.Base.<init>(Base.java:186)
        at processing.app.Base.main(Base.java:135)
Caused by: java.lang.ClassNotFoundException: com.fasterxml.jackson.core.JsonProcessingException
        at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 2 more

facchinm, 2017-12-18:
Mmmmh, are you compiling the IDE by yourself or using a precompiled package provided by Arduino? In the latter case it's probably an installation/extraction problem. Try downloading the package again and check for corruption using the provided [checksums](https://downloads.arduino.cc/arduino-1.8.5.md5sum.txt).

slava8686, 2017-12-18:
I compile a package, at start gives it.what is necessary not a flood plain

facchinm, 2017-12-18:
Well, I bet you are missing some configurations for the classpath; while a bit out-of-date, the build instructions are here https://github.com/arduino/Arduino/wiki/Building-Arduino .
Once you have a JDK and  `ant` in place, simply cd into `build` and run `ant build`.
That's it :wink: 
If you want some further help, please help a topic on the [forum](forum.arduino.cc). Thnaks

closed 2017-12-18 by facchinm