arduino/Arduino #7032: Editor Row & column display

alanjai, 2017-12-18:
I have used the Arduino for several years now and love your IDE. I've been tutoring newbies for a year now and in my own coding I always separate the code from comments by a large space (tabs) the way we did in ASM many years ago. Most of the newbies really like it because it "makes it look less complicated" as they can't yet visualize them separate . Would it be possible to add the column with the row indicator at the bottom of the window? It's very common in other editors and also be a huge help in layout for old guys like me _(and maybe the young too_). Thank you.

liamkinne, 2017-12-18:
This is a good suggestion, and if someone wants to spend the time adding that: go for it. The fact that the line number is already there, means it should display the column as well.

I understand for newbies that setting up an external text editor is overhead that doesn't need to be there for learning, but the use of a powerful external editor like Sublime Text is really rewarding. In fact my suggestion to you would be to use an external text editor.

per1234, 2017-12-18:
Duplicate of https://github.com/arduino/Arduino/issues/4403

closed 2017-12-18 by per1234

alanjai, 2017-12-18:
Firstly I wish to apologize to the developers for my search for duplicates not being thorough enough and hope I did not waste to much of your time. I am impressed by the quick response! I'll be more diligent next time. The suggestion to use an external editor is good advice indeed.

I myself will use an external editor for my more advanced projects but I want my student's to remain within the Arduino IDE  _to keep things simple_ so I prefer to remain within the IDE _with them_ for simplicity and consistency with the younger ones. The older students get introduced to the external editor solution.
Finally. Congrats to all of you hard workers for maintaining the Arduino effort! Thank you.