arduino/Arduino #7033: How can I use a single platform.txt for two of my custom boards.

haarislabs, 2017-12-18:
I have a custom board which has an Atmega 328 and esp module on it. I don't require any platform.txt for Atmega because the arduino ide uses the default path for it and for esp there is a specific platform.txt file.
Now I wanted to merge these two into a single file and to do that I have to have a single platform.txt file which defines the path for both the esp and Atmega. How can I do that???? can I use marcos such as #ifdef and tell the platform.txt to run that particular code when esp is selected and do nothing when Atmega is selected?

facchinm, 2017-12-18:
Hi @haarislabs , you can do it but, since the platforms are VERY DIFFERENT, you would end up with all the declarations in `boards.txt`, while `platform.txt` would be a simple placeholder for these declarations to expand at runtime.
For example, take a look at how `build.usb` is passed from `boards.txt` to the compiler commandline, and apply that strategy simply for everything.
Probably using two platforms is a better idea since you can (and should) inherit most stuff from pre-existing cores (to avoid useless duplication).

closed 2017-12-18 by facchinm

haarislabs, 2017-12-20:
What I wanted to do was to merge it into one single file. I'm avoiding duplication by setting the path to default arduino folder. For example for build.core=arduino:arduino .
So this avoids duplication. Now I was able to do what you told me and implement it but the problem now which occurs is since in the Esp module it has different sub-categories on the "Flash frequency", "Cpu Frequency" etc, the atmega is also using these configurations by default which creates an error. I considered creating a different sub-categoary which will only deal with esp parameters but I couldn't create it. Is there any way I can define and if/else kind of case where if the esp is selected then all those options are shown and hidden otherwise ?

per1234, 2017-12-20:
This issue tracker is only to be used to report bugs or feature requests. It is not appropriate to post here requesting support for your project. For that you should use the Arduino Forum, where you already have a thread on this topic (http://forum.arduino.cc/index.php?topic=517931). I'm sure we'll be able to help you with your problem over there.

haarislabs, 2017-12-21:
Hey @facchinm , can you just help me with one last thing? I wanted to change the default path which the Arduino sets for the gcc compiler when `#include` is used. I know by default it goes to the library directory but I wanted to know how is that path defined and how can I see it. Also, what would I have to do if I wanted to change it. 
I know this is used to report bugs but I've opened an issue in arduino as well but everyone just keeps ignoring me :(