arduino/Arduino #7034: Upload using programmer from another architecture gives java exception

benlye, 2017-12-18:
I have created boards definitions for a couple of boards used in a project I'm working on.  There are two avr boards but each needs it's own recipes in platform.txt, so I have created two architecture folders, let's say they are `avr` and `avr2`. The board package installation looks broadly like this:
```
project
    |-->hardware
        |--> avr
            |-->boards.txt
            |-->platform.txt
        |--> avr2
            |-->boards.txt
            |-->platform.txt
```

I would like to be able to use AVR programmers (USBasp, AVRISP, etc.) from either of my boards, but I am only able to use a programmer when I'm using the hardware board which is in the `avr` folder.  I can rename the folders so `avr2` becomes `avr` and vice-versa, and still only the `avr` folder works.  For the non-avr folder I get this error when I try to Upload Using Programmer:
```
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:314)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:89)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportAppHandler.run(Editor.java:2265)
	at java.lang.Thread.run(Thread.java:748)
```

The same thing happens when attempting to burn the bootloader.

It seems like the problem is related to the `avr2` folder name not matching the architecture of the Arduino avr boards, so the programmers are unknown.

So, my question is this: can I call a programmer from another architecture? 

Is this a bug in the IDE or a known limitation?  Either way, it would be nice if that error were friendlier.  It took me quite a while to work out that it was the hardware folder name causing the problem.

If it's not possible to reference a programmer from a different architecture, how should I package two avr boards which have different build recipes?  Can I do that in a single platform.txt file (which would allow me to put both boards in a single folder named `avr`?

cmaglie, 2017-12-18:
You should not use the architecture name `avr2` because such arch doesn't exists.
A lot of libraries refers to the architecture `avr` you should use only that one.

> how should I package two avr boards which have different build recipes?

Which part of the recipe is different? Maybe you can define a variable that you can set in boards.txt (for each board) and use in platform.txt something like the VID/PID for example:

https://github.com/arduino/Arduino/blob/cd9a15ebd1647336200b9328bc0912495e3a7195/hardware/arduino/avr/boards.txt#L39
https://github.com/arduino/Arduino/blob/cd9a15ebd1647336200b9328bc0912495e3a7195/hardware/arduino/avr/platform.txt#L130

benlye, 2017-12-18:
> You should not use the architecture name avr2 because such arch doesn't exists.

Understood.  Don't make up arch names.

> Which part of the recipe is different? 

The compiler flags and upload commands are different.  If I can override those from boards.txt then I think I can solve it and put everything in one package.

closed 2017-12-19 by per1234