arduino/Arduino #7038: Improvements for Stream.cpp in methods "parseInt" and "parseFloat"

fcebron, 2017-12-19:
I have noticed that the 2 methods "parseInt" and "parseFloat" in the [Stream.cpp](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.cpp) file which returns "0" when they reach a timeout. My problem with this is that I wanted to parse a String and find numbers (including zeros!) and to easily get read of characters which are not numbers.

To solve this, I suggest that these methods returns "(int)NAN" or "(float)NAN" in case that they reaches a timeout.

cousteaulecommandant, 2017-12-26:
There's no such thing as (int)NAN though.  Closest thing I can think of is `LONG_MIN` (-0x80000000), which is an unlikely value.

Maybe a better way around would be to add an optional parameter `int *successful` which "returns" 1 on success and 0 on error (similar to what `strtol()` does), or make additional `parseInt_check` and `parseFloat_check` functions that "return" the result in a pointed-to parameter and return 1 or 0 depending on their success status (similar to what `scanf()` does).

Ben-Rheinland, 2018-01-14:
This is related to issue #5011.

ScruffR, 2018-01-31:
This is also related to https://github.com/arduino/Arduino/issues/7177