arduino/Arduino #7041: [BUG] isWhitespace() / isSpace()

171231, 2017-12-21:
Good Morning facchinm
Issue #7039 not a BUG?
Fail!

the isWhitespace() function bring Char \9 (TAB) and \32 (SPACE) as true
the isSpace() bring \9 \10 \11 \12 \13 \32 as true

the ctype.h have isBlank() function,
Implementation?
Highligth isBlank() NONE / but isBlank() gives \9 and \32 as true

Also repair:
Arduino isWhitespace() -> ctype isspace()
Ardunio isSpace() -> ctype isblank() 

> facchinm
> This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

in https://www.arduino.cc/reference/en/language/functions/characters/isspace/
isSpace() true by ASCII 9 and 32 (DEC).
in https://www.arduino.cc/reference/en/language/functions/characters/iswhitespace/
isWhitespace() true then ASCII-Code 9, 10, 11, 12, 13 and 32 (DEC)

```
void setup()
{
  Serial.begin(9600); // Serial Init
  while (!Serial); // Wartet auf seriellen Monitor
}

void loop() 
{
  for (int thisChar = 0; thisChar < 128; thisChar++)
  {
    Serial.print("Char: \'");
    Serial.write(thisChar);
    Serial.print("\'  ASCII: ");
    Serial.print(thisChar);
    Serial.print("  ");
    if (isSpace(thisChar))            { Serial.print(", isSpace()"); }
    if (isWhitespace(thisChar))       { Serial.print(", isWhitespace()"); }
    Serial.println(".");
  }
  while(1);
}
```

This Code maen: isSpace(thisChar); is true by 9, 10,11, 12 13 32 (DEC) and maen isWhitespace(thisChar) is true by 9 and 32 (DEC)

matthijskooijman, 2017-12-21:
If you do not agree with closing an issue, it is usually better to comment on that issue rather than opening a new one. However, since you've opened this one, let's continue talking here. Also, watch your tone ("Fail!").

It is not clear to me (and apparently not @facchinm) what you think is the bug here. Your original post contained a sketch with the question "What is wrong here?", suggesting that you are looking for help with your sketch, even though the rest of the message looks like you're pointing out a bug.

Please clarify what you think is the bug exactly.

171231, 2017-12-21:
Good morning,

i'm wrote in #7039 :

> in Reference 
> https://www.arduino.cc/reference/en/language/functions/characters/isspace/
> **isSpace() true by ASCII 9 and 32 (DEC).**
> in https://www.arduino.cc/reference/en/language/functions/characters/**iswhitespace/
> **isWhitespace() true then ASCII-Code 9, 10, 11, 12, 13 and 32 (DEC)**
> 

and i wrote:
> This Code maen: **isSpace(thisChar); is true by 9, 10,11, 12 13 32 (DEC)** and maen **isWhitespace(thisChar) is true by 9 and 32 (DEC)**

this is not a problem with code - the code is for all to see the problem.
I have more than one this function and result checked.
also: what is wrong?

facchinm has closed this issue and i have search in the ctype.h.
Result is my new issue.

ok - "fail" was chosen "unhappy".

matthijskooijman, 2017-12-21:
It's hard to follow your English, but I'll try. Are you saying that isSpace and isWhitespace are swapped compared to the documentation?

> https://www.arduino.cc/reference/en/language/functions/characters/isspace/
isSpace() true by ASCII 9 and 32 (DEC).
in https://www.arduino.cc/reference/en/language/functions/characters/**iswhitespace/
isWhitespace() true then ASCII-Code 9, 10, 11, 12, 13 and 32 (DEC)

This looks correct: `isWhitespace` should probably return true for more characters than `isSpace`.

> This Code maen: isSpace(thisChar); is true by 9, 10,11, 12 13 32 (DEC) and maen isWhitespace(thisChar) is true by 9 and 32 (DEC)

This looks reversed.

Is this what you mean?

171231, 2017-12-21:
> This looks reversed.
> Is this what you mean?

YES! :-)
i have test from IDE 1.0.6 - 1.8.5
in all cases the same (wrong) result comes out.

if I use the c-function isblank() and isspace() test, the result is correct
But in the IDE is ` isblank() `not highlight.
Also is the isSpace() (Arduino) with isspace() (ctype.h) connected and thus a mistake in the implementation.
in the ctype.h documentation is the isspace()-funktion:
_**int isspace ( int __c )**
Checks for white-space characters. For the avr-libc library, these are: space, form-feed
(’ \ f’), newline (’ \ n’), carriage return (’ \ r’), horizontal tab (’ \ t’), and vertical tab (’ \ v’)._

and isblank():

_**int isblank ( int __c )**
Checks for a blank character, that is, a space or a tab._

my hard school English is more than 40 years ago - please excuse...

matthijskooijman, 2017-12-21:
> my hard school English is more than 40 years ago - please excuse...

No problem (if you also excuse our misunderstanding :-p)

It seems you're right, the implementation uses libc `isspace` to implement Arduino `isSpace`, but as you quoted from the manpages, the terminology is different.

https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/cores/arduino/WCharacter.h#L68-L72

https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/cores/arduino/WCharacter.h#L120-L124

So, certainly a bug, and the fix is simple: Just swap the contents of these functions (and fix the comments above the functions too). Probably also needs to happen in the SAM and SAMD cores, though.

171231, 2017-12-21:
I take it sporty...
OK.

The problem is not so much the code - there is a revised from the original translated into German reference and there are the descriptions and examples to vote.

If I know this is fixed, only one hint comes in - done.

Thank you!

matthijskooijman, 2017-12-21:
> The problem is not so much the code

I think it is: We could change the documentation, but then IMHO the functions do not make any sense.

I am not sure what you mean with the German reference?

171231, 2017-12-21:
> I think it is: We could change the documentation
Oh no.
isWhitespace() See: https://en.wikipedia.org/wiki/Whitespace_(programming_language) 
(in english an all over Language...)
But if I did not misunderstand you, it would fall on every C programmer's feet.

> I am not sure what you mean with the German reference?
I am currently helping to read, write and expand a nice documentation
https://www.arduinoforum.de/code-referenz

per1234, 2019-05-17:
Closing as fixed by https://github.com/arduino/reference-en/pull/595

The official decision was made to change the documentation rather than the functions to avoid a breaking change to the API:
https://github.com/arduino/ArduinoCore-API/pull/27#issuecomment-490122782

---
Regarding a reference in the German language, there is now an official German translation of the Arduino Language Reference:
https://www.arduino.cc/reference/de

closed 2019-05-17 by per1234