arduino/Arduino #7043: [Library Manager] Request to add "iBotX" Library

narinaviation, 2017-12-21:
This is a request to add the "iBotX" Library  from: https://github.com/narinaviation/iBotX

This library is to be used in the "I-BOT CAMP" of Robot Club, Faculty of Engineering
King Mongkut's Institute of Technology Ladkrabang (KMITL)

The camp is about learning the basics of programming for a robot to do some missions.
our target is high-school students from around the country.

The library may be too specific, it's because in the past (I-BOT 1 to I-BOT 10) our old version of the library was too hard to learn in 4 days of our camp, so our programming team decided to simplify the library to get it easier and more student-friendly. so they can learn programming in the more easier way.

"Anyway this is 100% non-profit and really for educational use."

So, We would appreciate if you could add this library to the Arduino's Library Manager.
so the students can download this in the easy way.

Best Regards...
Narin Bannasan
Programming Team Leader (PTL).
KMITL Robot Club, Faculty of Engineering,
King Mongkut's Institute of Technology Ladkrabang (KMITL)

cmaglie, 2017-12-22:
Done!

closed 2017-12-22 by cmaglie

narinaviation, 2018-01-05:
Thank you very much. ^_^