arduino/Arduino #7044: [Library Manager] Request to add Homeyduino Arduino library

renzenicolai, 2017-12-21:
Hello,

We've created a library which allows users to integrate their own Arduino projects with Homey (a central hub for a smart home). By installing the library in the Arduino IDE and installing the open-source Homeyduino app on a Homey the user can control Homey from an Arduino and vice-versa.

A set of [example projects](https://github.com/athombv/homey-arduino-library/tree/master/examples/Example%20projects) and [documentation](https://github.com/athombv/com.athom.homeyduino) can be found on the repositories of the app and library.

We hope you are willing to add our library to the Arduino Library Manager so that our users can easily download and update the library without having to download zip files from our Github repository.

Both the Homeyduino app and library are licensed with a GPLv3 license.

Thank you in advance!

Greetings,
Renze Nicolai

per1234, 2017-12-21:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

renzenicolai, 2017-12-21:
I have added a release as requested. Thank you for your pull request fixing the highlighting!

cmaglie, 2017-12-22:
Done!

closed 2017-12-22 by cmaglie