arduino/Arduino #7046: Broadcast Bonjour/Rendezvous more often

dirkx, 2017-12-23:
On mac - it may be an idea to broadcast the Bonjour/Rendezvous solicitation more often.

Perhaps every time the 'Tools' menu is opened.

That way we can be sure that the `Port' sub menu is populated after a (re)program.

facchinm, 2018-01-08:
Hi @dirkx ,
would you mind testing IDE 1.9.0 beta? We merged the updated jmdns which should be MUCH  more reliable at discovery (and removal notification)

per1234, 2018-07-26:
@dirkx did you ever get a chance to try out the beta build of the Arduino IDE? You can download it here:
https://www.arduino.cc/en/Main/Software#beta

sandeepmistry, 2019-09-16:
I'm closing this for now due to lack of feedback. Please re-open with requested info if needed.

closed 2019-09-16 by sandeepmistry