arduino/Arduino #7048: Close a tab

LouisPi, 2017-12-24:
I see no way to simply close a tab within the Arduino IDE! It makes me close all my tabs or delete the file!

per1234, 2017-12-24:
Closing as duplicate of https://github.com/arduino/Arduino/issues/2883

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2017-12-24 by per1234

LouisPi, 2017-12-24:
But there is still no answer to the question anywhere!