arduino/Arduino #7051: baudrate always back to 9600

jandemna, 2017-12-25:
For the Arduino Editor/Monitor I always have to set it to 115200 baud. It would be good if the baudrate could be set in the preferences menu.

per1234, 2017-12-25:
Is this regarding the [Arduino Web Editor](https://create.arduino.cc/editor)?

jandemna, 2017-12-25:
Yes, I always use the high baudrate and every time I start I have to adjust it.

per1234, 2017-12-26:
According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, we have to use the [Create > Editor section of the Arduino forum](http://forum.arduino.cc/index.php?board=101.0) for requesting feature requests or reporting bugs related to the Arduino Web Editor.

I have created a post for your feature request here:
http://forum.arduino.cc/index.php?topic=518949
so  I can now close this issue. Please feel free to continue the discussion on the forum.

closed 2017-12-26 by per1234