arduino/Arduino #7052: HTTP POST Parser Library

NatanBiesmans, 2017-12-25:
Created a new library that allows people to easily grab POST data from HTTP packets send from browsers, applications, .... 

This allows you to easily control the arduino via any webbrowser.

Examples added are the following:
- Simple light switch. Control an LED via the browser
- RGB controller. LED strip controller that can be controlled via any browser. (pc, mac, iphone, android, ...)

https://github.com/NatanBiesmans/Arduino-POST-HTTP-Parser

per1234, 2017-12-25:
Is this a request that your library be added to the Library Manager index? If so, some changes are required to meet the specifications detailed in the [Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ):
- Move the library to the root of the repository, rather than having it in the `Library` subfolder. You can either use the [Arduino Library 1.5 format](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files) or the [Arduino Library 1.0 format](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#old-library-format-pre-15).
- Add a library.properties file, formatted according to the [specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format).
- Create a [release ](https://help.github.com/articles/creating-releases/)or [tag ](https://git-scm.com/book/en/v2/Git-Basics-Tagging)that matches the `version` value in your library.properties file.

After doing so, please comment here so we know we can proceed.

NatanBiesmans, 2017-12-25:
Hi Per1234,

Thanks for the quick response. I indeed seek to add the library to the arduino Library Manager Index.

I adjusted the file layout and added the required files as per your recommendation.

Kind regards,
Natan Biesmans

cmaglie, 2017-12-30:
Done!

closed 2017-12-30 by cmaglie