arduino/Arduino #7055: Client.h irregularities

jandrassy, 2017-12-27:
The class `Client` inherits from `Stream` which inherits from `Print`. 
`Client` declares pure virtual methods that are already declared in `Print` (`write(uint8_t)`) and `Stream` (`available()`, `peek()`, `read()`). 
Class `Client` for no good reason declares as pure virtual methods implemented in `Print` (`flush()`,  `write(buffer, size)`.  The `flush()` method is here out of context of printing/transmitting without a comment and for that reason  implemented wrong in some Client implementations (WiFiClient) as clearing of receive buffer.
And `Client` adds a duplicate method for reading into a buffer. In `Stream` `readBytes(bufer, size)` is implemented using method `read()`. `Class.read(buffer, size)` is pure virtual.
If some implementation of `Client` is handled as `Stream`, `readBytes(buffer, size)` is used and not the 'forced' implementation of `read(buffer, size)`.
Some implementations of `Client` may implement `read(buffer, size)` calling `readBytes(buffer, size)`, so now  `readBytes(buffer, size)` can't be added as inline for `read(buffer, size)`.

facchinm, 2018-01-08:
Hi @jandrassy ,
would you mind providing some POC to show how the issue can be solved (while keeping the backwards compatibility with existing user code) ? Thanks

jandrassy, 2018-01-08:
ok. first I remove the duplicate declarations and try compiling all implementations I can find and I can run Ethernet2, WiFiEsp and WiFi Link examples.
the `read(buffer, size) vs. readBytes` can only be resolved if inspection of all major implementations  shows that they do not implement read as 'alias' to `readBytes`. then `readBytes` could by added to Client as 'alias' to perhaps optimized `read(buffer, size)`.

jandrassy, 2018-01-14:
so, I think I have it. Overriding readBytes in Client.h with call to read(buff, size) could cause a disaster. The read(buff, size) method is rarely used and the implementations may have bugs as I found out with WiFi Link. Clean solution is marking read(buf, size) somehow as deprecated and give it a default implementation calling readBytes.
Following version of Client.h has no impact on existing code. And it better integrates Client.h in the Print/Stream hierarchy and warns about the read(buf, size) method.
```
#ifndef client_h
#define client_h
#include "Print.h"
#include "Stream.h"
#include "IPAddress.h"

class Client : public Stream {

public:
  virtual int connect(IPAddress ip, uint16_t port) =0;
  virtual int connect(const char *host, uint16_t port) =0;
  virtual void stop() = 0;
  virtual uint8_t connected() = 0;
  virtual operator bool() = 0;

  /**
   * \deprecated use readBytes
   * Client.read(buf, size) is a historical method of class Client,
   * left for backward compatibility.
   * If Client is accessed as Stream, this method is not visible.
   */
  virtual int read(uint8_t *buf, size_t size) {
	return readBytes(buf, size);
  }

protected:
  uint8_t* rawIPAddress(IPAddress& addr) { return addr.raw_address(); };
};

#endif

```

jandrassy, 2018-04-30:
@facchinm, sorry, I now discovered that readBytes is one of timed methods of Stream - uses timedRead to fill the buffer. So Client adds a not timed read to buffer (to get the contents of internal buffer).  Perhaps read(buf, size) should be added to Print, with default implementation. If not then in Client the read method should have a comment similar to this. (And no implementation as it is in current Client.h.)

/**
* Client.read(buf, size) is a method of class Client,
* If Client is accessed as Stream, this method is not visible.
*/
  virtual int read(uint8_t *buf, size_t size) = 0;