arduino/Arduino #7056: Need to download Arduino 1.0.5-r2 for Mac OSx but I can't find it

mheimbinder, 2017-12-28:
I believe the files use to be available via this link, https://downloads.arduino.cc/arduino-1.0.5-r2-macosx.zip, but it no longer works.  Do you know where I can download these files?

cmaglie, 2017-12-28:
Old releases are archived here: https://www.arduino.cc/en/Main/OldSoftwareReleases#1.0.x

http://downloads.arduino.cc/arduino-1.0.5-macosx.zip seems what you are looking for.

mheimbinder, 2017-12-28:
Thanks so much for responding.  

Unfortunately, I'm not looking for 1.0.5 but 1.0.5-r2 for Mac OSx. I can find the release on the "Release Notes" page here, https://www.arduino.cc/en/Main/ReleaseNotes, but it is not present on the "Previous IDE Releases" page you linked to.  Any idea where I might download 1.0.5-r2 for Mac OSx.  Note that I have asked this same question in the Arduino Forums, see http://forum.arduino.cc/index.php?topic=519313.0

Thanks in advance for your assistance.

cmaglie, 2017-12-28:
The release `1.0.5-r2` has been made only for Windows since it has fixed the Windows drivers (at the times the driver signature expired because it wasn't "timestamped" so we were forced to re-release the 1.0.5), as you can see from the release notes:

```
ARDUINO 1.0.5-r2 - 2014.01.08

* Signed drivers for Windows 8.1
* Fixed Windows drivers signature (that prevented installation on
  some Windows 8.x OS). Now the signature is timestamped and should
  not expire.
```

So, 1.0.5 and 1.0.5-r2 are **exactly** the same, except for the windows drivers signature, that's why we didn't re-release it for macosx and linux.
Since it caused this kind of confusion we stopped making `-r2` releases and make only full release even if only change is a single small file.

mheimbinder, 2017-12-28:
Now I understand.  Thanks for the clarification.  It is much appreciated.

closed 2017-12-28 by mheimbinder