arduino/Arduino #7057: Need to download Arduino 1.0.5-r2 for Mac OSx but I can't find it

mheimbinder, 2017-12-28:
Thanks so much for responding cmaglie.

Unfortunately, I'm not looking for 1.0.5 but 1.0.5-r2 for Mac OSx. I can find the release on the "Release Notes" page here, https://www.arduino.cc/en/Main/ReleaseNotes, but it is not present on the "Previous IDE Releases" page you linked to. Any idea where I might download 1.0.5-r2 for Mac OSx. Note that I have asked this same question in the Arduino Forums, see http://forum.arduino.cc/index.php?topic=519313.0

Apologies, for opening a new issue but I wasn't sure if a response inside a closed ticket would be responded to.  If I'm not following best practices please let me know.  I'm new to this repository so advice is appreciated.

Thanks in advance for your assistance.

per1234, 2017-12-28:
Closing as duplicate of https://github.com/arduino/Arduino/issues/7056

closed 2017-12-28 by per1234

matthijskooijman, 2017-12-28:
@mheimbinder, just add more comments to your previous issue, rather than opening new issues and cluttering the list.

mheimbinder, 2017-12-28:
Thanks for the tip matthijskooijman.  Will do.