arduino/Arduino #7061: Repeat affixed

chenzengpeter, 2017-12-29:
I am using arduino on mac but will paste it again when I copy it.

per1234, 2017-12-29:
It's unclear what you're saying. You need to provide much more detail.

chenzengpeter, 2017-12-30:
if i (command c) ---> copy  print
but (command v) ---> Posted on the print will be repeated affixed twice  It becomes printprint

per1234, 2017-12-30:
Which version of the Arduino IDE are you using? Please test with the [beta build](https://www.arduino.cc/en/Main/Software#beta) to verify the problem has not already been fixed.

I seem to remember this having been reported previously but I can't find it. I actually did have this same problem for a while but haven't for a long time and I can't remember what changed. I do have a keyboard remapping of my copy/paste shortcuts and changed the application I use to do that at one point so it may have been caused by a bug in the previous application.

chenzengpeter, 2017-12-30:
I currently use Arduino 1.8.5
I tested the problem still exists
And then you see what you say changed the shortcut key inside the thing
I would like to change (command v) to (control v or option v) one of the shortcut keys
Fantastic thing happened (command v) can be used normally  but press (control v or option v) can not be used
A magical bug
I remember when I bought it (command v) can be used normally to update from a mac version to die 
i buy mac in 2014 year
thank you for your help

lukec, 2018-07-30:
This bug happens when your keyboard input is set to anything other than US English. It's easy to reproduce on a mac:

1. set your keyboard input to a different locale (eg: Dvorak or other languages)
2. open any sketch 
3. copy some text
3. paste (Cmd-V) into the sketch.

Expected: pasting "foo"

> foo

Actual:

> foofoo

Sidchou, 2018-10-18:
same problem, basically the txt editor read my short cut key twice when  i do it once, also happen with Cmd+/, it comment out and then back the line

NeilFraser, 2019-06-26:
This bug dates back to at least April 2015, version 1.6.3:
https://forum.arduino.cc/index.php?topic=315092.0

It is *extremely* irritating.  Of course since it mostly affects non-Americans, I guess it's pretty low on the priority list...

gweep, 2020-03-03:
I have this problem still.... here is my solution..   PASTE (cmd-v)  gets you two copies.... UNDO (cmd-z) removes the 2nd copy only....   inelegant but it works for me.

airs0urce, 2020-03-21:
I have same issue

wtrout, 2020-10-14:
I'm a DVORAK user who just switched to mac. Dang this bug is annoying.

auxiliaryjoel, 2021-04-06:
does anyone have a soluation for this. I screenshot items to my clipboard constantly as part of training for my clients,
in the past pasting this content behaved as it should - it would only paste the screenshot once into my gmail emails.
But in the last month or so that has changes to now always pasting the screenshots 2 at a time. 
What on earth has changed? (Mac OS 10.14.6 on Imac)

prog666, 2021-07-07:
same for me, and I thought it has nothing to do with Dvorak I use because I tried Russian to check - but appears they are both broken...

mertnesvat, 2022-06-21:
yep it's related to the keyboard layout in Dvorak it's double pasting but USA keyboard works fine.

gweep, 2022-06-22:
Holy cow!  How long is this going to remain unaddressed?

per1234, 2022-06-22:
There is a good chance it already has been addressed. Please check to see whether it still occurs with the latest nightly build of Arduino IDE 2.x:

https://www.arduino.cc/en/software#nightly-builds

zaneclaes, 2023-05-20:
@per1234 if you look just above your post, you'll see that someone tested the nightly build just 4 days before your post. Some of us have been [repeatedly reporting this issue for over 3 years now](https://github.com/arduino/Arduino/issues/10534). 

The _problem has actually gotten MUCH worse after the latest nightly build and the switch to Arduino IDE_. Now, the hotkeys do something completely different. For example, if I press `CMD+C` (using Dvorak+QWERTY) to copy something, it instead makes the Output tab disappear (and does not copy anything). If I press `CMD+V` (to paste), it says `CMD+K was pressed, waiting for more keys` (and again does not actually do anything).