arduino/Arduino #7063: LTC2633 library

TamojitSaha, 2017-12-31:
LTC2633 - Dual 12-/10-/8-Bit I2C VOUT DACs with 10ppm/°C Reference

This library demonstrates the usage of the DAC with Arduino over i2c communication protocol.

per1234, 2017-12-31:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

TamojitSaha, 2017-12-31:
Updated keywords.txt

TamojitSaha, 2017-12-31:
v1.0.0 release for LTC2633 deployed!

https://github.com/TamojitSaha/LTC2633/releases/tag/v1.0.0

TamojitSaha, 2018-01-01:
v1.0.1 [release](https://github.com/TamojitSaha/LTC2633/releases/tag/v1.0.1) for LTC2633 deployed! 
Please update the changes.

per1234, 2018-01-01:
You must update the `version` value in library.properties before creating the release. Because you did it after the release, the `version` value in your 1.0.1 release is still set to 1.0.0:
https://github.com/TamojitSaha/LTC2633/blob/v1.0.1/library.properties#L2
Your library will be added to the Library Manager index after some days and from then on any new releases you create will be automatically added to the Library Manager index shortly after that so there is no need to notify us of them.

TamojitSaha, 2018-01-01:
I am closing this issue and applying for new one.

SandeepanSengupta, 2018-01-01:
The udated library is released [here](https://github.com/TamojitSaha/LTC2633/releases/tag/0.1.0)

TamojitSaha, 2018-01-02:
@per1234  Please comment if there is anything that is not right

per1234, 2018-01-02:
Here's what I notice:
https://github.com/TamojitSaha/LTC2633/blob/master/library.properties#L1
```
name= miniDAC
```
but the library specification says
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>- **name** - the name of the library. Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). ***It cannot start or end with a space***, and also it cannot start with a number.

That makes me think the space before the name is a problem.

SandeepanSengupta, 2018-01-02:
@per1234 , please find the corrected library [here](https://github.com/SandeepanSengupta/miniDAC-library/releases/tag/0.1.0)

per1234, 2018-01-02:
@SandeepanSengupta that tag is in your fork of the repository. Unless you want to change the request to add the https://github.com/SandeepanSengupta/miniDAC-library repository instead of https://github.com/TamojitSaha/LTC263, you will either need to create the 0.1.0 tag in https://github.com/TamojitSaha/LTC2633. Please let me know how you want to proceed.

SandeepanSengupta, 2018-01-02:
Please add the following repo
https://github.com/SandeepanSengupta/miniDAC-library

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Regards,
Sandeepan Sengupta

TamojitSaha, 2018-01-02:
@per1234 , Is there anything else we need to perform ?

SandeepanSengupta, 2018-01-02:
@per1234 , Please index the [this](https://github.com/SandeepanSengupta/miniDAC-library) repo and this is the latest release [(v0.1.0)](https://github.com/SandeepanSengupta/miniDAC-library/releases/tag/0.1.0)

cmaglie, 2018-01-08:
Done, thanks!

closed 2018-01-08 by cmaglie

TamojitSaha, 2018-01-08:
Checked! Working...! Thanks a lot!

per1234, 2019-01-05:
The Library Manager indexer checks all the libraries on the list for new tags/releases hourly. So you don't need to make a request every time you release a new version; they'll be picked up automatically. In fact, your 1.0.0 already has!

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list