arduino/Arduino #7071: Add a "latest" link to downloads.arduino.cc

jamessynge, 2018-01-02:
It would be convenient to be able to download the latest version of the Arduino IDE/tools via a fixed URL for use from a Travis Continuous Integration build. For example, [the Software page](https://www.arduino.cc/en/Main/Software) gets updated to link to the latest release by number (e.g. 1.8.5 today), but that isn't easy for a script to figure out.

Please add a link that is updated each time there is a new release. For example, instead of these two links:
```
https://www.arduino.cc/download_handler.php?f=/arduino-1.8.5-linux64.tar.xz
https://downloads.arduino.cc/arduino-1.8.5-linux64.tar.xz
```
It would be most convenient if these links existed:
```
https://www.arduino.cc/download_handler.php?f=/arduino-latest-release-linux64.tar.xz
https://downloads.arduino.cc/arduino-latest-release-linux64.tar.xz
```

per1234, 2018-04-30:
@lmihalkovic are you asking whether it display a notification when a newer IDE version is available? If so, the functionality is in place (controlled via File > Preferences > Check for updates on startup) but it hasn't worked for a very long time due to a webpage not being updated (https://github.com/arduino/Arduino/issues/5496).

It's not clear to me how that could be relevant to this topic.

lmihalkovic, 2018-04-30:
@per1234 really? would that stable link not be usable for a immediate download, especially for eg if the server would respond to a http HEAD call?

fedekrum, 2019-04-16:
I think that what the Op is asking for is to have a symbolic link to the latest-releases with a unique URL,

For example.
I have made a scripts that tries to install latest Arduino IDE to a newly installed raspberry pi.
My script first figures out id the OS is 32 or 64 bit. The it should download via curl the LATEST ARM version for it.

As the links of the files are usually
`https://downloads.arduino.cc/arduino-1.8.9-linuxarm.tar.xz`
that will be a problem to know the release number and automate the download.

But if there is a symbolic link of it to
`https://downloads.arduino.cc/arduino-latest-release-linuxarm.tar.xz`
to download the same file, it will do our life much easier !!

As this world tends to automate stuff, that will help a lot.