arduino/Arduino #7073: Can't upload skecth with desktop Arduino but with Arduino Create YES

warriorfenixSM, 2018-01-02:
Hi,

I have some time make a probes because is it Arduino desktop not important that version (installer, portable, 1.6.5, 1.8, nighly). Can't upload a simple sketch in my board but when i try to upload a sketch with Arduino web editor, the code works and burn my arduino.

The board is: Arduino Zero
SO: Windows 10.

Thanks

per1234, 2018-01-02:
Please do this:
1. File > Preferences > Show verbose output during: > compilation (uncheck) > upload (check) > OK
1, Sketch  > Upload
1. After the upload fails you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
1. Paste the error messages in a reply here in a code block (`<>` button on the toolbar).

warriorfenixSM, 2018-01-02:
C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp: In member function 'void DallasTemperature::setHighAlarmTemp(const uint8_t*, char)':

C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:600:25: warning: comparison is always false due to limited range of data type [-Wtype-limits]

     else if (celsius < -55) celsius = -55;

                         ^

C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp: In member function 'void DallasTemperature::setLowAlarmTemp(const uint8_t*, char)':

C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:616:25: warning: comparison is always false due to limited range of data type [-Wtype-limits]

     else if (celsius < -55) celsius = -55;

                         ^

C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp: At global scope:

C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:781:6: warning: unused parameter 'deviceAddress' [-Wunused-parameter]

 void DallasTemperature::defaultAlarmHandler(const uint8_t* deviceAddress){}

      ^

Archiving built core (caching) in: C:\Users\User\AppData\Local\Temp\arduino_cache_102981\core\core_arduino_samd_mzero_bl_6548007a8f973269f1850a5583ec367c.a
El Sketch usa 21320 bytes (8%) del espacio de almacenamiento de programa. El máximo es 262144 bytes.
Forzando reinicio usando 1200bps apertura / cierre en el puerto COM3
PORTS {COM1, COM3, } / {COM1, } => {}
PORTS {COM1, } / {COM1, } => {}
PORTS {COM1, } / {COM1, COM5, } => {COM5, }
Found upload port: COM5
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -v -patmega2560 -cstk500v2 -PCOM5 -b57600 -Uflash:w:C:\Users\User\AppData\Local\Temp\arduino_build_892527/Simple.pde.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : stk500v2
         Overriding Baud Rate          : 57600
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
[... 22 lines cut ...]
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_recv(): programmer is not responding
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

cmaglie, 2018-01-02:
You should first install the support for the Arduino Zero:

- Open "Tools" menu -> "Boards" -> "Boards Manager..."
- On the new opened window write "Zero" in the search box -> click in the middle of "Arduino SAMD Boards" item in the bottom part -> click "Install" button.
- Once the install completes (it takes some minutes), close the window
- Open "Tools" menu -> "Boards" -> select "Arduino Zero (Programming Port)"

This should be enough to upload successfully.

Anyway this issue tracker is to **report bugs** on the IDE or the libraries.
If you need help to get started you should read the official documentation https://www.arduino.cc/en/Guide/ArduinoZero or ask for help in the forum https://forum.arduino.cc

closed 2018-01-02 by cmaglie