arduino/Arduino #7081: HyperLink Error

agamagtdM10102148, 2018-01-04:
the link: 
Download a DFU Programmer
This is the software that will update the code 
Windows: Download Atmel's flip programmer from here......

Where "here" should be navigated to the following (Because Atmel had been bought by Microchip):
http://www.microchip.com/developmenttools/productdetails.aspx?partno=flip

krgoodwin, 2018-01-14:
I just went through the process of updating the MEGA8U2 on an Uno with Windows 7 and the version of FLIP hosted on microchips website does not include drivers for Arduino Uno DFU nor does it support AT90USB82 at all form what I can tell.

I ended up using dfu programmer 0.7.2 which included the correct drivers. I wouldn't bother updating the link to a program that doesn't work but instead update the page to a process that works.

(https://forum.arduino.cc/index.php?topic=522536.0)

https://sourceforge.net/projects/dfu-programmer/files/dfu-programmer/0.7.2/

william-s-owen, 2018-01-15:
It appears that they have, for windows, version 2.4.6 only.

krgoodwin, 2018-01-16:
There are multiple references (forums posts, videos) to Windows FLIP 3.4.x from a couple years ago. It looks like Microchip is hosting the old 2.4.6 version which doesn't include the necessary drivers.

william-s-owen, 2018-01-23:
Looks like they put up 3.4.7 for windows on http://www.microchip.com/developmenttools/productdetails.aspx?partno=flip

krgoodwin, 2018-01-23:
Wow, I was not expecting them to actually do anything helpful like that. Looks like my previous comments can be disregarded. Just need a link update on the original page now.

kengdahl, 2018-02-19:
Fixed

closed 2018-02-19 by kengdahl