arduino/Arduino #7082: Bad avrdude path in arduino-PR-beta1.9-BUILD-31 on Linux

jeffbuhrt, 2018-01-04:
This seems to be a repeat of an older bug like: https://github.com/arduino/Arduino/issues/4110

java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:796)
	at processing.app.SketchController.exportApplet(SketchController.java:767)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2241)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more
An error occurred while uploading the sketch

Laptop used: Fedora 27, 64bit Intel.

Workaround change Downloads/arduino-PR-beta1.9-BUILD-31/hardware/arduino/avr/platform.txt
from:
tools.avrdude.path={runtime.tools.avrdude.path}
to:
tools.avrdude.path=/home/YourUser/Downloads/arduino-PR-beta1.9-BUILD-31/hardware/tools/avr/
and uploads now work.

facchinm, 2018-01-08:
Hi @jeffbuhrt , it is indeed the revamp of that old bug, but this time the fix is there so it could be a bit more complicated. Could you launch the IDE from the commandline with `--get-pref` flag and paste here the results?

jeffbuhrt, 2018-01-08:
@facchinm I put arduino-PR-beta1.9-BUILD-31/hardware/arduino/avr/platform.txt back to the original tools.avrdude.path before running this:

$ ./arduino-PR-beta1.9-BUILD-31/arduino --get-pref
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
#PREFDUMP#
last.sketch.count=2
update.check=false
platform.auto_file_type_associations=true
gui.scale=auto
editor.window.width.default=500
editor.window.height.default=600
editor.window.width.min=400
editor.window.height.min=290
editor.code_folding=false
editor.font=Monospaced,plain,12
editor.antialias=true
editor.external=false
editor.caret.blink=true
editor.invalid=false
editor.linenumbers=false
editor.keys.alternative_cut_copy_paste=true
editor.keys.shift_backspace_is_delete=true
editor.keys.home_and_end_beginning_end_of_doc=false
console=true
console.output.file=stdout.txt
console.error.file=stderr.txt
console.lines=4
console.auto_clear=true
console.length=500
editor.tabs.expand=true
editor.tabs.size=2
editor.indent=true
editor.auto_close_braces=true
editor.divider.size=0
editor.save_on_verify=true
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.display=1
export.application.platform=true
export.application.fullscreen=false
export.application.stop=true
export.applet.separate_jar_files=false
export.delete_target_folder=true
preproc.save_build_files=false
preproc.color_datatype=true
preproc.web_colors=true
preproc.enhanced_casting=true
preproc.substitute_floats=true
preproc.substitute_unicode=true
preproc.output_parse_tree=false
preproc.imports.list=java.applet.*,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.*,java.net.*,java.text.*,java.util.*,java.util.zip.*,java.util.regex.*
browser=mozilla
run.present.bgcolor=#666666
run.present.stop.color=#cccccc
run.present.exclusive=false
target_package=arduino
target_platform=avr
board=nano
software=ARDUINO
build.warn_data_percentage=75
programmer=arduino:avrispmkii
upload.using=bootloader
upload.verify=true
serial.port=/dev/ttyUSB0
serial.databits=8
serial.stopbits=1
serial.parity=N
serial.debug_rate=115200
editor.languages.current=
runtime.ide.path=/home/jbuhrt/Downloads/arduino-PR-beta1.9-BUILD-31
runtime.ide.version=10900
boardsmanager.additional.urls=
build.verbose=true
compiler.cache_core=true
compiler.warning_level=default
custom_cpu=nano_atmega328
editor.update_extension=true
last.ide.1.8.4.daterun=1514232026
last.ide.1.8.4.hardwarepath=/usr/share/arduino/hardware
last.ide.1.9.0-beta.daterun=1515428735
last.ide.1.9.0-beta.hardwarepath=/home/jbuhrt/Downloads/arduino-PR-beta1.9-BUILD-31/hardware
last.screen.height=1050
last.screen.width=3046
last.serial.location=211,163,1155,407
last.sketch.default.location=433,84,500,600,395,0
last.sketch.default.path=/home/jbuhrt/Arduino/Blink2.arduino/Blink2.arduino.ino
last.sketch0.location=1786,5,839,978,773,0
last.sketch0.path=/home/jbuhrt/Arduino/Blink2.arduino/sketch_stepper2/sketch_stepper2.ino
last.sketch1.location=1493,73,500,829,624,0
last.sketch1.path=/home/jbuhrt/Arduino/Blink2.arduino/Blink2.arduino.ino
last.sketch2.location=432,60,780,697,411,0
last.sketch2.path=/home/jbuhrt/Arduino/Blink2.arduino/Blink2.arduino.ino
launcher=xdg-open
proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=auto
recent.sketches=/home/jbuhrt/Arduino/Blink2.arduino/sketch_stepper2/sketch_stepper2.ino,/home/jbuhrt/Arduino/stepper_oneRevolution/stepper_oneRevolution.ino,/home/jbuhrt/Arduino/Blink2.arduino/Blink2.arduino.ino
serial.port.file=ttyUSB0
serial.port.iserial=null
sketchbook.path=/home/jbuhrt/Arduino
upload.verbose=false
runtime.os=linux
runtime.platform.path=/home/jbuhrt/Downloads/arduino-PR-beta1.9-BUILD-31/hardware/arduino/avr
runtime.hardware.path=/home/jbuhrt/Downloads/arduino-PR-beta1.9-BUILD-31/hardware/arduino

drinkypoo, 2018-01-26:
Same problem on Windows 7, Arduino 1.6.5 installed in the regular location, beta version dropped into c:\programs (are there instructions for using the beta? they are not near the download link)

I was asked to put version of avr core reported by library manager into the PR, but loading library manager, looking at installed packages, and searching for both "avr" and "core" did not turn up anything like an avr core in either version, so I don't know what that was supposed to mean.

facchinm, 2018-01-29:
Hi @drinkypoo ,
I need you to check the AVR core version on "Board Manager", not "Library manager" (you can find it at the top of "Tools/Board" submenu. Thanks!

matthijskooijman, 2019-09-19:
I was also running into this error when installing a git version of the AVR core directly into my sketchbook and deleting the bundled AVR core, see #9237. I do suspect that @jeffbuhrt was not doing any such things, so this is *probably* a different issue, but I wanted to link it just in case it is helpful.