arduino/Arduino #7087: A Thermostatic radiator valve for arduino?

Stevemoretz, 2018-01-06:
Hello people, I was wondering what to buy for doing such a thing the point is I don't want the valve to be that smart to control the temperature of the room all I want is to let the arduino take care of the temperature by a temperature sensor connected to the arduino and some codes the only thing that valve should do is to rotate when the arduino orders is to but all valves I see are smarter and do everything themselves so there's expensive and not capable of being controlled by an arduino (I don't have Bluetooth on my arduino just wifi(esp8266)).What should I do?I can't find anything good. Please guide me Tnx.

per1234, 2018-01-06:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-01-06 by per1234

Stevemoretz, 2018-01-06:
Sorry I counted the TVR as a motor please move the topic to the appropriate
part I'm so sorry.