arduino/Arduino #7089: the Launcher icon looks like Processing

teo1978, 2018-01-06:
[image]

The attached screenshot shows what the Arduino IDE's Launcher icon on Ubuntu looks like.

I understand that the IDE is based on Processing, but I think it deserves its own icon. Even the tooltip says "Processing [random version]".

facchinm, 2018-01-08:
Hi  @teo1978 ,
this is super strange. Which IDE version are you running? Did you install it via `install.sh` in the release tarball?

teo1978, 2018-01-13:
Actually, I didn't run install.sh at all. I just extracted the .tar.gz (note there are no installation instractions whatsoever in the download page) and noticed there was an executable called "arduino", so I launched it, and it worked.

I do have Processing 2.2.1 (I'm not sure I ever "installed" that one either), so maybe at some point some "association" has been made between whatever binary is at the core of both and that icon. In my opinion there must be something deeply wrong either in the way Ubuntu manages applications or in some or both of these two particular applications for this to happen, even assuming I did something wrong.

facchinm, 2018-01-15:
It is surely related with files association and installed Processing (probably because the main executable is still called `pde.jar` in Arduino too). The Processing icon is not part of the IDE resources so it's likely to be overridden by Ubuntu itself. Should be reported upstream at some point, closing it here since we can't do much :slightly_smiling_face:

closed 2018-01-15 by facchinm