arduino/Arduino #7094: Update some libraries name

fpistm, 2018-01-08:
Hi,
Following this PR #6752, all libraries from STM32duino GitHub organization are prefixed with "STM32duino" in the library.properties name field.

Those following libraries have been added before this naming "guideline":
   - https://github.com/stm32duino/SPBTLE-RF
   - https://github.com/stm32duino/MX25R6435F
   - https://github.com/stm32duino/LSM6DS3
   - https://github.com/stm32duino/LSM6DS0
   - https://github.com/stm32duino/LSM303AGR
   - https://github.com/stm32duino/LPS25HB
   - https://github.com/stm32duino/HTS221
   - https://github.com/stm32duino/LSM6DSL
   - https://github.com/stm32duino/LPS22HB
   - https://github.com/stm32duino/STM32SD
   - https://github.com/stm32duino/STM32Ethernet

So displayed name in the Arduino Library Manager is not aligned with library.property name field.
Moreover, when one of those libraries is installed it appears twice in the list (one without the prefix and one with it).

Thanks in advance

cmaglie, 2018-01-15:
I've changed the library names, the indexer should publish the result in a couple of minutes.

Should I do the same for:
https://github.com/stm32duino/LwIP.git

?

fpistm, 2018-01-15:
Thanks Cristian.
Yes you can. I forgot it.

closed 2018-01-15 by cmaglie