arduino/Arduino #7097: Arduino IDE fails to start under windows if "Documents" directory is unavailable.

WestfW, 2018-01-09:
Early in its initialization, the Arduino cod calls Win32KnownFolders.getDocumentsFolder, which does a catch-less:
    return new File(Shell32Util.getFolderPath(ShlObj.CSIDL_MYDOCUMENTS));

If the Documents folder is not accessible, this fails and throws an exception as shown by arduino_debug (see attached backtrace.)

This can be easily reproduced by creating a windows user and modifying their documents folder to be on a flash drive, ejecting the flash drive, and then trying to start Arduino.  It can also occur if the Documents directory is configured to be on a network share that is unreachable.

Because this happens so early in the initialization, it fails even if the IDE has been installed in "portable" mode, which should prevent it from needing to access the Documents directory.  (Thus: no known workaround.)

It looks like other code is prepared to deal with a null sketchbook path; perhaps all that is needed is ignoring the error...

WestfW, 2018-01-09:
Traceback:

C:\Users\breakme\Downloads\arduino\arduino-1.8.5>.\arduino_debug.exe
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Looking for library 'kernel32'
Adding paths from jna.library.path: null
Trying kernel32.dll
Found library 'kernel32' at kernel32.dll
com.sun.jna.platform.win32.Win32Exception: The system cannot find the path specified.
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:48)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:61)
        at cc.arduino.os.windows.Win32KnownFolders.getDocumentsFolder(Win32KnownFolders.java:71)
        at processing.app.windows.Platform.recoverDefaultSketchbookFolder(Platform.java:76)
        at processing.app.windows.Platform.init(Platform.java:55)
        at processing.app.Base.<init>(Base.java:201)
        at processing.app.Base.main(Base.java:144)

facchinm, 2018-01-10:
Hi @WestfW ,
I pushed 76fa511 to beta branch to overcome this issue. The build should be available soon.

per1234, 2018-01-16:
From the forum thread (http://forum.arduino.cc/index.php?topic=521015.msg3559240#msg3559240)
WestfW said:
>I can confirm that the latest. Beta code did fix my artificially broken terms system just fine

No word from the user who originally reported the issue yet.

per1234, 2019-09-16:
The original reporter of the bug has reported to me that the problem is solved (via a PM on the forum) and I see the fix is merged into the `master` branch, so I'm going to close this as solved.

closed 2019-09-16 by per1234