arduino/Arduino #7098: Arduino IDE 1.8.5 overrides baud rate specified in board.txt file (set to 115200bps)

dphamminh, 2018-01-09:
When trying to use Arduino IDE 1.8.5 with a sanguinololu board, I noticed that the baud rate specified in board.txt file is not taken into account.

The Arduino IDE always sets the baud rate to 115200 instead of taking what is specified in board.txt file (57600).

As my sanguinololu board only supports 57600 bps, so I could'nt upload the firmware with Arduino 1.8.5. 

The feature is working on Arduino IDE 1.0.1.

per1234, 2018-01-09:
You'll need to provide enough information for us to reproduce the issue.
- Instructions for how to install the hardware support package
- Exact selections for the **Tools > Board** and any other custom Tools menus

The hardware package I'm familiar with is:
https://github.com/Lauszus/Sanguino

If you examine the boards.txt file you'll see that the upload baud rate depends on your selection in the **Tools > Processor** menu.

If I select **Tools > Processor > ATmega644 or ATmega644A (16 MHz)** the Arduino IDE uploads at 115200, as would be expected
https://github.com/Lauszus/Sanguino/blob/master/avr/boards.txt#L28:
```
sanguino.menu.cpu.atmega644.upload.speed=115200
```
If I select **Tools > Processor > ATmega644 or ATmega644A (8 MHz)** the Arduino IDE uploads at 57600, also as expected
https://github.com/Lauszus/Sanguino/blob/master/avr/boards.txt#L40:
```
sanguino.menu.cpu.atmega644_8m.upload.speed=57600
```

matthijskooijman, 2018-01-09:
> The Arduino IDE always sets the baud rate to 115200 instead of taking what is specified in board.txt file (57600).

Where do you see this? Do you mean 115200 is passed to `avrdude` during the upload? That could be a bug. Or do you mean this baudrate is used in the serial monitor? That is by design, since the baudrate in boards.txt is used for uploads, not communication.

dphamminh, 2018-01-09:
The Arduino environment is used to control a sanguinololu board with ATMega1284 processor at 16 MHz. The application is the control of a 3D printer : Geeetech Aluminium Prusa I3. 

The firmware for the printer is here :
https://www.geeetech.com/forum/download/file.php?id=1651&sid=d8b76b9ece6689ff9c7b818f4a38e637

The sanguinololu environment is downloaded from :
https://www.geeetech.com/forum/download/file.php?id=3604&sid=d8b76b9ece6689ff9c7b818f4a38e637

I have changed the speed in board.txt to set it to 57600 bps. In the Arduino environment, I select the sanguinololu board and the ATMega1284 processor with 16 MHz. 

When I try to download the firmware, I got the following message from the IDE:
Arduino : 1.8.5 (Windows 7), Carte : "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe1
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x71
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xb0
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xf1
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x26
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x98
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xb0
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x49

avrdude done.  Thank you.

I think that the uploading problem comes from the fact that the upload speed used (115200) by avrdude is not the one expected by my board (57600).

dphamminh, 2018-01-09:
PS : When I use the same files with arduino 1.0.1, the speed is correct and the uploading successful

matthijskooijman, 2018-01-09:
Try enabling verbose upload in the preferences, then you can see the exact speed passed to avrdude.

per1234, 2018-01-09:
>The sanguinololu environment is downloaded from :
https://www.geeetech.com/forum/download/file.php?id=3604&sid=d8b76b9ece6689ff9c7b818f4a38e637

If that's the one you're actually using with Arduino IDE 1.8.5 then you must have made some other significant changes since it's not compatible with >1.0.x IDE versions.

Please provide the exact files you're using.

dphamminh, 2018-01-09:
Please find the files that I use. I have slightly modified the Prusa Firmware, to solve the compatibility issue. 

The Arduino zip file contains the sketch environment with the sanguinololu board.txt description

[Prusa_I3_A1.1.zip](https://github.com/arduino/Arduino/files/1616576/Prusa_I3_A1.1.zip)
[Arduino.zip](https://github.com/arduino/Arduino/files/1616579/Arduino.zip)

per1234, 2018-01-09:
Both the Sanguino and Sanguino_old hardware packages you have installed are in Arduino 1.0 format. They will not be recognized by the Arduino IDE. You must have another version of the Sanguino hardware package installed elsewhere on your system. This is why editing those files has no effect.

It's now clear this is a support issue rather than a bug. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.
3. Add a link to this issue report in your forum post so we will have access to the information that has already been uncovered.

closed 2018-01-09 by per1234

dphamminh, 2018-01-10:
Dear per1234, 

You were right. I was editing the wrond board.txt configuration. In fact I used the sanguinololu board manager : https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json . It automatically instals the sanguinololu hardware in the following directory : C:\Users\<username>\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.2

When editing the right board.txt file, the baud rate setting is correct and I can use the Arduino 1.8.5 IDE.

Thanks a lot for your support.

Duc

jimSchaeferEnsign, 2020-04-09:
Another point about changing the baud rate settings in board.txt.... make sure you close out ALL instances of the Arduino IDE before editing.  I edited boards.txt while the IDE was open and  the change never appeared.  Only after I closed the IDE, edited boards.txt then reopened the IDE was I able to get the baud rate to 'take'. 

I also experienced the "avrdude: stk500_getsync()" error when trying to download to an Atmega 1284P that was on a Tronxy produced Meltzi 2.0_V5 board. The the Sanguino 1.0.3 1284 16 MHZ board selection defaulted to 115200,  I was using Arduino V1.8.9 IDE. When I changed the baud rate to 57600 in boards.txt it worked well.  The clue was that I was able to download using the 8MHZ  selection which was set to 57600. However, this didn't work. Octopi would not connect, the baudrate the serial monitor connected was twice what I programmed in the firmware (which makes sense now).