arduino/Arduino #7102: [Library Manager] Add WiFiManager Library

jakerabid, 2018-01-09:
It'd be great if I could get this added to the library manager index:

https://github.com/jakerabid/WiFiManager

Thanks!

per1234, 2018-01-09:
@tzapu's [WiFiManager library](https://github.com/tzapu/WiFiManager) (which yours is a fork of) is already in the Library Manager index. You will need to change the `name` value in your library.properties file to a unique value and then create a new release/tag or redo the previous release/tag.

Please also be sure the `version` value in your library.properties file matches the release version, which is not the case with your current 0.14 release:
https://github.com/jakerabid/WiFiManager/blob/0.14/library.properties#L2:
```
version=0.13
```

Once you have done this please comment here so we will know we can proceed with adding your library to the Library Manager index.

jakerabid, 2018-01-09:
Thanks for your patience @per1234 - I'll get that tidied right up.

jakerabid, 2018-01-09:
@per1234 I've tidied up the library.properties file, and created a new tag.

cmaglie, 2018-01-23:
Done

closed 2018-01-23 by cmaglie