arduino/Arduino #7103: Error downloading Infineon .json url in Arduino IDE

webstone13, 2018-01-10:
Has anyone programmed XMC microcontroller by Infineon Technologies using Arduino IDE?

I tried as per the steps given in the GitHub repo of Infineon. Below is the link of the same:

[https://github.com/Infineon/XMC-for-Arduino/wiki/Implementation-in-Arduino-IDE](url)

After I copied the .json url in the preferences, I am getting following error:

http://downloads.arduino.cc/packages/package_index.json file signature verification failed. File ignored.

Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino

What can be a possible solution for it? I am running Arduino IDE 1.8.5 on Windows 10. Is this has anything to do with it?

facchinm, 2018-01-10:
Hi @webstone13 ,
the problem here is that http://downloads.arduino.cc/packages/package_index.json gets ignored due to a certificate error. I am sure that the certificate is ok so the problem must be somewhere else. 
Which IDE version are you running? You can try removing manually the cached jsons in
* `C://Users/youruser/AppData/Arduino15/` if you are on Windows
*  `/Users/youruser/Library/Arduino15/` if you are on Mac
*  `~/.arduino15/` if you are on Linux

and restart the library manager.

webstone13, 2018-01-10:
Hi @facchinm,
I am running Arduino IDE 1.8.5 on Windows 10. What are cached jsons? Are they same as .tmp file in Arduino15 folder. If yes, I have already tried that and it didn't work.

Thanks for the help.

webstone13, 2018-01-10:
@facchinm 
Didn't work. Please find the .json file in the link below.
[https://github.com/Infineon/Assets/releases/download/current/package_infineon_index.json](url)

facchinm, 2018-01-10:
Sorry if I was not clear enough, the problem seems related with Arduino-provided json (the default one called `package_index.json`). I need you to attach a zip containing the (json and sig) files that are bing downloaded to your PC (in the Arduino15 folder). Thanks

webstone13, 2018-01-12:
[json files.zip](https://github.com/arduino/Arduino/files/1624993/json.files.zip)

Dear @facchinm 
Please find attached all the .json files in Arduino15 folder.

facchinm, 2018-01-12:
Does your Arduino15 folder only contained that files? No other .json or .sig? I'm starting to think that it could be a problem related with a very strict antivirus running in the background and deleting random files it thinks could harm the system. Try adding the Arduino15 folder to its whitelist and/or check if the needed files (namely  `package_index.json` and `package_index.json.sig`) has been quarantined.

webstone13, 2018-01-12:
Dear @facchinm 
Thanks for your help. I tried doing the same in standlone version of Arduino IDE and it works perfectly fine there.
I don't know what the problem was in the original version but somehow it is working now.

What you are suggesting maybe true. It may be possible as I am trying this on my office PC which quite secured.

Thanks anyway.

facchinm, 2018-01-12:
I'm closing the issue in the meantime, changing the title so it may be easier to find if someone else encounters it. Thanks for your time.

closed 2018-01-12 by facchinm

jhedup, 2018-02-09:
Please open the issue again. I get this message on two different computers one on version 1.8.4 and one on 1.8.5: "http://downloads.arduino.cc/packages/package_index.json file signature verification failed. File ignored." 

I deleted the 1.8.5 version's Arduino15 directory and reinstalled the IDE a few times without success.

ankitdaf, 2018-02-11:
+1
Getting the same issue as above, both on windows and linux. This should likely be an issue with the file signature on the file online.

Unable to work with any board,and the existing boards that I had added got deleted as well

facchinm, 2018-02-12:
@jhedup @ankitdaf I just retested it from a clean installation and the signatures are ok. So the problems could be related with:
- a temporary network error on our download server
- an aggressive antivirus trying to stop the download/tamper the file in some way so the signature is incorrect
Could you retry deleting all Arduino15 folder content right now and check if the problem still applies? Thanks

sheltoncyril, 2018-02-17:
Still facing the same issue, cannot download SAM boards core because of this. I think most users are facing this. I'm trying to setup my IDE to program an STM32 board but it is dependent on the SAM core, I had to use arduino 1.6.5 as it ignores the file signature verification. Updating it to the latest version doesn't work as it uninstalls the previous version and it starts throwing errors on the console. Even this youtuber I follow has this error shown in one of the videos he made :p https://youtu.be/EaZuKRSvwdo?t=2m31s here's the link to it. Anyway I did find a work around, by installing the portable version of arduino 1.6.5 then copying the tools from that to arduino15 in  appdata/local. It still throws errors but I am able to program my STM32 this way. Hope to find a fix soon thx.

sheltoncyril, 2018-02-18:
Ok I think something was wrong with my JAVA installation and my router. I restarted my router without PPPOE and setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true pasted this in CMD. Now it's working fine, but i'm really not sure of what was actually causing it. I doubt my Pi-Hole had anything to do with this but it could be a reason for it.

jhedup, 2018-02-18:
@facchinm Apologies for the slow response, I am just a weekend warrior. I uninstalled Arduino IDE, deleted Arduino15 folder and manually deleted most reference to Arduino in the Windows 10 registry. I still found the same problem after installing 1.8.5. I will now test over a different network and let you know.

jhedup, 2018-02-18:
Connected via my mobile ISP without on both my note books, one on 1.8.4 left as is as control and the one on 1.8.5 with the fresh install. No success on both. Next up is to check the java settings and turn off anti virus.

sheltoncyril, 2018-02-18:
Why don't you use 1.6.5r5 temporarily till a permanent fix is found? I'm trying my best to figure it out but my main suspicion is my internet or PPPOE. Was hoping that the issue would come back so that I can find out the exact cause and it did :p after a reboot it started throwing the same errors again, downloading the latest version of java didn't help.
So give me a few hours i'll try and figure a fix for this.

sheltoncyril, 2018-02-18:
Okay it is definitely a network issue. When I switched over to my mobile network, it immediately started working, but on wifi it keeps throwing the same error. Probably related to PPPOE as disabling that helped me download on my home internet. Will get back with more info soon.

jhedup, 2018-02-18:
@sheltoncyril Thanks for the update. I tried your Java setting also did not help. In my case changing the network had not solved the problem. I will use the 1.6.5r5 as you suggested in the mean time.

jhedup, 2018-02-24:
@facchinm Thank you for fixing! During the week the Boards Managers complained about a file error but now both my fresh 1.6.5.r.5 and my untouched 1.8.5 works.

sheltoncyril, 2018-02-24:
Sorry I couldn't get back, was kinda busy with my exams :D hope I helped you. @facchinm the issue seems to be fixed now, would like to know why this was occuring.

KamranArain, 2018-08-09:
I was also facing same issue like signature error. I just connected My PC with my sim card data network and try to download and it works for me. that was network issue that's why it was not working. Thanks for your support.

red-scorp, 2019-06-22:
Same here with 1.8.9. Can it be IPv6 issue?

per1234, 2019-06-22:
@red-scorp there's a newer issue report for this error which might be more relevant to your situation: https://github.com/arduino/Arduino/issues/8988

You can provide the information requested on that thread to assist with the investigation.

chrisly42, 2019-07-07:
@red-scorp You were right!

After hours of unsuccessful attempts, changing the run.options to

run.options=-Djava.net.preferIPv4Stack

in preferences.txt fixed the issue for me.

red-scorp, 2019-07-07:
@chrisly42 unfortunately it does not help in my case

bplondon, 2020-03-29:
Allowing access for arduino.exe in Windows 10 sec. did the job.

hosenmatz, 2020-03-29:
hello together,
today I was facing the same "signature problem" in Windows 7 for version 1.8.12
solution was to start arduino.exe with administrative rights
hope that helps

H

EdwinRB, 2020-03-30:
Hi to all.
Same problem today.
Windows 7 Professional SP1
Please help.
Edwin

EdwinRB, 2020-03-30:
> 
> 
> Hi to all.
> Same problem here today, IDE 1.8.12, starting arduino.exe with or without administrator privileges
> Windows 7 Professional SP1
> Please help.
> Edwin

Solved!.  I did nothing.  I believe that the error was caused by the restrictions, filtering and blocking of the Internet that we have in my country ;(
Regards

playaspec, 2021-07-29:
Experiencing the same issue under Ubuntu 20.04, Arduino 1.8.15 installed from the tar file. Ran the install script for local install, then straight to the library manager. The IDE itself displays the error:

`https://downloads.arduino.cc/libraries/library_index.json.sig file signature verification failed. File ignored.`

I run no AV, am not behind NAT (machine directly on a public IPv4 address), and have no firewall rules that could impact operation.  From the output in the console, it appears that MD5 validation is failing. I have attached all the output in a zip file.

[ArdError.gz](https://github.com/arduino/Arduino/files/6902081/ArdError.gz)

The file: `~/.arduino15/cache/cache.json` reads as follows:

`{
  "files" : [ {
    "remoteURL" : "https://downloads.arduino.cc/libraries/library_index.json.sig",
    "localPath" : "/home/playaspec/.arduino15/cache/downloads.arduino.cc/libraries/library_index.json.sig",
    "eTag" : "cb2d350a22052a61dafc54fbd47c7796",
    "md5" : "MD5:cb2d350a22052a61dafc54fbd47c7796",
    "createdAt" : "2021-07-29T12:35:50.983",
    "cacheControl" : {
      "maxAge" : 14400,
      "mustRevalidate" : false,
      "noCache" : false,
      "noStore" : false
    },
    "expires" : "2021-07-29T16:35:50.983"
  }, {
    "remoteURL" : "https://downloads.arduino.cc/libraries/library_index.json",
    "localPath" : "/home/playaspec/.arduino15/cache/downloads.arduino.cc/libraries/library_index.json",
    "eTag" : null,
    "md5" : null,
    "createdAt" : "2021-07-29T12:35:51.152",
    "cacheControl" : {
      "maxAge" : 14400,
      "mustRevalidate" : false,
      "noCache" : false,
      "noStore" : false
    },
    "expires" : "2021-07-29T16:35:51.152"
  }, {
    "remoteURL" : "https://downloads.arduino.cc/libraries/library_index.json.gz",
    "localPath" : "/home/playaspec/.arduino15/cache/downloads.arduino.cc/libraries/library_index.json.gz",
    "eTag" : "f9c1b4cc8618c2562a012ae1abe66ba4",
    "md5" : "MD5:f9c1b4cc8618c2562a012ae1abe66ba4",
    "createdAt" : "2021-07-29T12:36:41.202",
    "cacheControl" : {
      "maxAge" : 14400,
      "mustRevalidate" : false,
      "noCache" : false,
      "noStore" : false
    },
    "expires" : "2021-07-29T16:36:41.202"
  } ]
}
`

The entry for `library_index.json` has a NULL md5. I downloaded both the .sig and .gz manually, and verified the md5sums match. I then tried editing the cache.json, adding the md5sums I generated from the `library_index.json` file I downloaded and ran again. It again complained about the `library_index.json.sig` file not passing the hash check. It rewrote the `cache.json` file, removing the `library_index.json.gz` entry completely, and resetting the `library_index.json` back to NULL values for both the eTag and MD5.

Is it possible that the `library_index.json` file on the CDN is corrupt or tampered with? How can this deadlock be resolved? I'd really like to get on to the task at hand, but with the number of supply side malware attacks of late, this has me a little spooked.

Could someone please do their due diligence and verify that content on the Arduino's CDN is trustworthy, and explain the root cause of this ongoing problem once it is resolved?

------------------------------------------------------
About to hit `comment` when I reviewed the comments one more time. I had tried adding the `run.options=-Djava.net.preferIPv4Stack` as @chrisly42 suggested to add to the `preferences.txt` to no effect. Then I noticed that @sheltoncyril had used `run.options=-Djava.net.preferIPv4Stack=true` which seems to have worked.

Hope this all helps.