arduino/Arduino #7121: [Library Manager] Add Plaquette library to the library manager

sofian, 2018-01-13:
Plaquette is a new Arduino library that turns your Arduino environment into an object-oriented environment.

First release now available tagged "v0.1"

https://github.com/SofaPirate/Plaquette

per1234, 2018-01-13:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

sofian, 2018-01-13:
I am sorry, I created the tag locally but did not push it properly.
Just pushed it as "v0.1" tag now: https://github.com/SofaPirate/Plaquette/tree/v0.1

cmaglie, 2018-01-15:
Done!

closed 2018-01-15 by cmaglie