arduino/Arduino #7122: [Library Manager] Add library with non semver versioning

gioblu, 2018-01-13:
Hi there, I have developed 2 libraries that are now used by many Arduino IDE users:
- https://github.com/gioblu/PJON
- https://github.com/gioblu/Cape

I would like to add them to the Arduino IDE list but I read here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated

Both libraries contains non semver release version tags.
Will PJON and Cape library list add request be denied because of that?
Tags change to existing releases in order to meet this requirement is quite painful.
Is enough to have the latest release following semver standard?

per1234, 2018-01-13:
From the library specification (https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format):
> Version should be semver compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

So you should be fine with your current versioning scheme.

gioblu, 2018-01-13:
@per1234 thanks for the response time near c (speed of light) 👍 
Will double check all the requirements and open an issue requesting the addition. 
Thank you very much again.

gioblu, 2018-01-13:
Will close here because the isse has been solved. Thank you again.

closed 2018-01-13 by gioblu