arduino/Arduino #7131: Please add Switch library to Library Manager

avandalen, 2018-01-16:
https://github.com/avandalen/Switch

This library is used for years now and is stable.
Supports also long press and double clicks.

avandalen, 2018-01-20:
Dear Per
Thank you, but I can't find it at the lib manager. Maybe something get wrong?

per1234, 2018-01-20:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2018-01-23:
Done

closed 2018-01-23 by cmaglie