arduino/Arduino #7136: [Library] A FABRIK 2D inverse kinematics solver in Arduino!

henriksod, 2018-01-18:
https://github.com/henriksod/Fabrik2DArduino

Example provided. This is a very useful library for solving inverse kinematics on a small system like an Arduino. Inverse kinematics is used for e.g. robotic arms to control the end effector position and orientation.

This library is a simplified inverse kinematics solver, allowing only 2D coordinates for the joints. A 3D version may be added in the future, but it would probably take up too much resources on the Arduino.

per1234, 2018-01-18:
I'm assuming this is a request for your library to be added to the Library Manager index. Correct? If so, there are a couple things you need to do before your library can be added:

Add a library.properties file to the root of the repository. You can find the description of the required format in the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2018-01-18:
library.properties looks good but you forgot to create the 0.1 release or tag.
More info:
- Release: https://help.github.com/articles/creating-releases/

OR

- Tag: https://git-scm.com/docs/git-tag
Remember you need to push the tag to GitHub after creating one on your local clone of the repo.

henriksod, 2018-01-19:
I forgot to push the tag. I have done it now!

per1234, 2018-01-19:
Your library.properties file is currently fine but at the time of the 0.1 tag there were several problems with it. Please either redo the 0.1 tag or bump the version value in library.properties and push a new matching tag.

henriksod, 2018-01-19:
I made a new tag and pushed it as 0.3. Sorry for the inconvenience

per1234, 2018-01-19:
The `version` value in library.properties was still set to 0.1 at the time of the 0.3 tag:
https://github.com/henriksod/Fabrik2DArduino/blob/0.3/library.properties#L2
It was only updated to 0.3 after the tag.

The `version` value should match the tag name, which means you need to update the value in library.properties, commit, then create the tag at or after that commit.

henriksod, 2018-01-19:
Okay, it is done now.

cmaglie, 2018-01-23:
Done

closed 2018-01-23 by cmaglie