arduino/Arduino #7137: Issue with Windows10 Installation

MrJenny, 2018-01-18:
I friend of mine has recommended my Teensy board. I think it is nice.

But now I am running in to a issue. I did try to install the Arduino IDE on my  1st Laptop.Win10 including all the drivers. The Arduino IDE does start but the communication does fail. I Spent 2h debugging with no success. 
The I tried on my 2nd Laptop.Win10 the Arduino doesn’t even start as Administered or not.
I NEED HELP! Please

per1234, 2018-01-18:
>I friend of mine has recommended my Teensy board. I think it is nice.
But now I am running in to a issue. I did try to install the Arduino IDE on my 1st Laptop.Win10 including all the drivers. The Arduino IDE does start but the communication does fail.

This issue tracker is only to be used to report bugs or feature requests. That topic is more appropriate for either the [PJRC Forum](https://forum.pjrc.com/forum.php) or the [Arduino Forum](http://forum.arduino.cc/). I'm sure you'll get help you with your problem over there.

>The I tried on my 2nd Laptop.Win10 the Arduino doesn’t even start as Administered or not.

This part might be appropriate for this issue tracker but we would need more information. Please do this:

1. Download the hourly build from https://www.arduino.cc/en/Main/Software#hourly
2. Unzip the downloaded file to a convenient location.
3. Run the file arduino.exe that you will find inside the unzipped IDE folder.
4. If the Arduino IDE doesn't start then start the Windows command line (`cmd`).
5. Run the file arduino_debug.exe, which you will find in the unzipped IDE folder.
6. After the IDE fails to start you will see some output printed in the command line window. Copy ALL the output.
7. Paste the output in a reply here in a code block by using the **`<>`** button on the toolbar.

per1234, 2018-01-18:
Are you using the Arduino IDE or Teensyduino?

MrJenny, 2018-01-18:
@per1234

Thanks for your support: Here is the result by using the arduino_debug.exe

`C:\Users\marcel\Desktop\Teensy\arduino-nightly>arduino_debug.exe
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
DPI detection failed, fallback to 96 dpi
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Looking for library 'kernel32'
Adding paths from jna.library.path: null
Trying kernel32.dll
Found library 'kernel32' at kernel32.dll
com.sun.jna.platform.win32.Win32Exception: Der Netzwerkpfad wurde nicht gefunden.
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:48)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java:61)
        at cc.arduino.os.windows.Win32KnownFolders.getDocumentsFolder(Win32KnownFolders.java:71)
        at processing.app.windows.Platform.recoverDefaultSketchbookFolder(Platform.java:76)
        at processing.app.windows.Platform.init(Platform.java:55)
        at processing.app.Base.<init>(Base.java:201)
        at processing.app.Base.main(Base.java:144)`

facchinm, 2018-01-18:
I believe it's related with https://github.com/arduino/Arduino/issues/7097
@MrJenny  would you mind testing the Beta version of the IDE and report if it also happens there? Thanks!

MrJenny, 2018-01-19:
buhaaa!!

The Beta Version of Arduino seems work but the Teensy loader can not be added. Even if cheat with the version.txt file.. 

Any hint how I can add the Teensy 3.5 things to Arduino IDE?

per1234, 2018-01-19:
The fact that the Beta version solves the startup issue shows that the bug you reported was previously reported and fixed so I'm closing this as a duplicate of #7097.

The Teensy issue is outside the scope of this issue tracker. Note that there are two possible solutions:
1. Fix the problem that is causing #7097. You can find an explanation of the cause in that issue report. We may be able to help you with that on the A[rduino forum](http://forum.arduino.cc/).
2. Figure out how to use Teensy with the beta version of the Arduino IDE. Note that IDE version is primarily intended for beta testing so you may encounter some newly introduced [bugs ](https://github.com/arduino/arduino-preprocessor/issues)when using it. Even so, a rough-around-the-edges IDE is better than none at all, right? For help with this option, I recommend searching the [PJRC forum](https://forum.pjrc.com/forum.php) to see if that topic has already been discussed and, if not, make a post about it there.

closed 2018-01-19 by per1234

nitish7134, 2018-10-29:
Same problem. I tried the portable version too, but Arduino IDE starts, then  initializes packages and just vanishes
the output on running debugger gave:
`Loading configuration...
Initializing packages...
Error parsing library.index:Unexpected character ('<' (code 60)): expected a valid value (number, String, array, object, 'true', 'false' or 'null')
 at [Source: (FileInputStream); line: 1, column: 2]
Preparing boards...
java.lang.NullPointerException
        at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:140)
        at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:109)
        at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:677)
        at processing.app.Base.onBoardOrPortChange(Base.java:1313)
        at processing.app.Base$11.actionPerformed(Base.java:1514)
        at processing.app.Base.rebuildBoardsMenu(Base.java:1490)
        at processing.app.Base.<init>(Base.java:267)
        at processing.app.Base.main(Base.java:145)`

PaulStoffregen, 2018-10-29:
@nitish7134 - Maybe you have an old version of Arduino?  That error sure looks like #7917 which was fixed in Arduino 1.8.7.

HarshaAbeydeera, 2019-05-13:
GUYS  I have the same problem I started with Administrator / Compatibility mods didn't work in  my WINDOWS  10 PC :( I can see the SPLASH SCREEN but it disappear quickly

PaulStoffregen, 2019-05-13:
> GUYS I have the same problem

Do you have the latest Arduino 1.8.9, or an older version?  Is it the original from Arduino, or has it been modified by installing Teensyduino?

If you really are running 1.8.9, look in the folder where it's installed (default on Windows is C:\Program Files (x86)\Arduino) for the arduino_debug.exe version.  Run that one, since it will give info in a console window that remains on your screen.  That info is critical to figuring out what's going wrong.

HarshaAbeydeera, 2019-05-15:
> Do you have the latest Arduino 1.8.9, or an older version? Is it the original from Arduino, or has it been modified by installing Teensyduino?
>
> If you really are running 1.8.9, look in the folder [...]

Thank You..but  I solved the issue Its my Antivirus (Windows Defender) :)

Nitheshnithu121, 2020-06-10:
I have same problem when i start  Arduino IDE it starts, then initializes packages and just vanishes......
The output on running debugger is as follows:
C:\Program Files (x86)\Arduino>arduino_debug.exe
Set log4j store directory C:\Users\user\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.main(Base.java:150)

Cloud any one please help me solve this problem

cmaglie, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

alimoal, 2020-06-10:
> We had a problem on our package_index.json that caused the error and prevented the IDE to start.
> Now the package_index has been fixed, but to recover a working installation you must delete once [...]

thank you!!!!!!!!! it worked for me

ilvento, 2020-06-10:
Hi,
Installing the latest Hourly Build (1.8.13 2020-06-03 04:33) fixed the problem for me.
Cheers
Ben

jwestmoreland, 2020-07-08:
Hello Everyone,

I wanted to add something - I was having this issue - Documents was of course 'available' - but - I had disabled OneDrive for various reasons - and after installing new versions of the IDE (for Portenta) - I discovered nothing was really working correctly until I re-enabled OneDrive.  I plan on 'fixing' this - but it was not apparent to me that OneDrive was the root cause nor did the fact one of my dev PC's appears to rely on OneDrive that's disturbing to me.... anyway - thought I'd mention this here in case it saves someone else from going down this goofy rabbit hole.

Note the IDE would just crash - same symptoms as in OP.

Regards,
John W.

chiduciot, 2021-02-28:
Loading configuration...
Initializing packages...
DEBUG StatusLogger Using ShutdownCallbackRegistry class org.apache.logging.log4j.core.util.DefaultShutdownCallbackRegistry
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger AsyncLogger.ThreadNameStrategy=UNCACHED (user specified null, default is UNCACHED)
TRACE StatusLogger Using default SystemClock for timestamps.
DEBUG StatusLogger org.apache.logging.log4j.core.util.SystemClock does not support precise timestamps.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Took 0.069349 seconds to load 209 plugins from sun.misc.Launcher$AppClassLoader@1e6f5c3
DEBUG StatusLogger PluginManager 'Converter' found 44 plugins
DEBUG StatusLogger Starting OutputStreamManager SYSTEM_OUT.false.false-1
DEBUG StatusLogger Starting LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@98ac89]...
DEBUG StatusLogger Reconfiguration started for context[name=1e6f5c3] at URI null (org.apache.logging.log4j.core.LoggerContext@98ac89) with optional ClassLoader: null
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger PluginManager 'ConfigurationFactory' found 4 plugins
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Missing dependencies for Yaml support, ConfigurationFactory org.apache.logging.log4j.core.config.yaml.YamlConfigurationFactory is inactive
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Not in a ServletContext environment, thus not loading WebLookup plugin.
DEBUG StatusLogger Using configurationFactory org.apache.logging.log4j.core.config.ConfigurationFactory$Factory@9159b
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.properties] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.properties] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.properties] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yml] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yaml] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yaml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yaml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.yaml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.json] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.json] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.json] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.json] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.jsn] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.jsn] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.jsn] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.jsn] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.xml] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.xml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.xml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test1e6f5c3.xml] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.properties] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
TRACE StatusLogger Trying to find [log4j2-test.properties] using ClassLoader.getSystemResource().
TRACE StatusLogger Trying to find [log4j2-test.yml] using context class loader sun.misc.Launcher$AppClassLoader@1e6f5c3.
TRACE StatusLogger Trying to find [log4j2-test.yml] using sun.misc.Launcher$AppClassLoader@1e6f5c3 class loader.
[... cut ...]
DEBUG StatusLogger Shut down OutputStreamManager SYSTEM_ERR.false.false, all resources released: true
DEBUG StatusLogger Appender Console stopped with status true
TRACE StatusLogger XmlConfiguration stopped 2 remaining Appenders.
TRACE StatusLogger XmlConfiguration cleaning Appenders from 2 LoggerConfigs.
DEBUG StatusLogger Stopped XmlConfiguration[location=jar:file:/D:/arduino/lib/pde.jar!/log4j2.xml] OK
DEBUG StatusLogger Stopped LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@98ac89] with status true