arduino/Arduino #7147: Arduino Web editor does not allow managing additional boards

jm-andonegi, 2018-01-21:
I am working with Controllino boards. When using IDE, I can use Additional Boards Manager URL and then I can select the Controllino model I want to use ([this is the URL](https://raw.githubusercontent.com/CONTROLLINO-PLC/CONTROLLINO_Library/master/Boards/package_ControllinoHardware_index.json) of Controllino boards)

In the web editor, I have not been able to find that feature. The editor recognises the board as "Arduino/Genuino Mega or Mega 2560" but I can not compile the sketch.

It would be great to have the "Additional Boards Manager URL" feature in the web editor

per1234, 2018-01-21:
According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, the [Create > Editor section of the Arduino forum](http://forum.arduino.cc/index.php?board=101.0) is the appropriate place to submit bug reports or feature requests related to the Arduino Web Editor.

This feature has already been requested a number of times there:
http://forum.arduino.cc/index.php?topic=361376

closed 2018-01-21 by per1234

jm-andonegi, 2018-01-21:
I was not able to find that forum section.

Thanks for your help! :)