arduino/Arduino #7152: Library in the library manager is renamed

avandalen, 2018-01-24:
This library in the lib manager
https://github.com/avandalen/Switch
can be deleted and this copy wich has become another name can be added
https://github.com/avandalen/avdweb_Switch

Note:
I prefer to add a unique name (e.g. avdweb) to the folder name. This may be recommended for all libraries? The advantage of this is that you always know the origin of the libraries.

per1234, 2018-01-24:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed.

GitHub does allow you to change the name of repositories via the Settings tab if you like, rather than creating a new repository. It will automatically redirect from the old URL but it would still be best to update the URL in the Arduino Library Manager index.

Note that you have changed the repository name but not the library name (as set via the `name` property in library.properties). That's fine; they don't need to match. I'm just making sure you're aware of this.

cmaglie, 2018-01-24:
> I prefer to add a unique name (e.g. avdweb) to the folder name. This may be recommended for all libraries? The advantage of this is that you always know the origin of the libraries.

this is the recommended way to do it, especially with very common names like `LED` or `Switch`, we always suggest to add a prefix (like your `avdweb_Switch`) or invent a unique name for the library.

It is also recommended to rename the include to match the library name, so you should rename also `Switch.h` to `avdweb_Switch.h`, this helps the IDE to correctly select your library (it may be that other libraries provides a `Switch.h` header and in that case the IDE need to choose among one of those).

BTW, as pointed out by @per1234, you should make all the changes (in particular the `name` field in `library.properties`), commit & push them, and re-do the tag **after** having pushed the changes.

avandalen, 2018-01-25:
I changed also the filenames to avdweb_Switch.h and avdweb_Switch.cpp

per1234, 2018-01-25:
At the time of the v1.0.4 release the `version` property was still set to 1.0.3:
https://github.com/avandalen/avdweb_Switch/blob/v1.0.4/library.properties#L2
It was only set to 1.0.4 after the release was created.

The `name` property's value in your library.properties remains Switch.

Once you have resolved these issues please comment here so we know we can proceed with your request.

avandalen, 2018-01-27:
The release is okey now. The library name is still Switch, because of it is used by many people. Just the filenames are renamed to avoid lib conflicts.

cmaglie, 2018-01-31:
Done

closed 2018-01-31 by cmaglie