arduino/Arduino #7154: ISSUE with USING WINUSB Driver with Arduino AS USB

newpie18, 2018-01-24:
Project Goal: Create a custom USBDevice.
Name of device : "CapiaZ"
Board USed: Arduino MICRO
summary:
I was able to create a unique boatloader for my CapiaZ board, it has a specific VID & PID for it. and when you plug it to the pc the ARDUINO IDE will show the port and the name "CAPIAZ".
Then I used WINUSB.sys and WINUSB.inf to generate the driver. Successfully, I am able to install the driver and Device manager shows it as follwoing -which is the goal so far-:

+CapiaZ Device
CapiaZ-S01
Also I wrote my C++ library that communicate with CapiaZ using WINUSB.lip and SETAPI.lib. Using read Endpoints bulik in and Builk out to read and write from it.
I use WinUsb_WritePipe and WinUsb_ReadPipe,. I write data to the Capia and read data from the capia no problems.

**_# The issue,_**
when I unplug (soft or hard) the Capia then plug it back, and try to run my C++ application, the C++ application will connect and write to it but I'm not able to read data from it. any suggestions what could be the issue ?
thanks

dlabun, 2018-01-24:
Since you are no longer using the Arduino bootloader (or even a clone of it) it's very difficult to tell what the problem is without access to your source code. This isn't really an Arduino issue so you're best bet is to ask for help on the forums.

closed 2018-01-24 by per1234

newpie18, 2018-01-25:
NO I didn't change the boatloader  I used the Caterina boatloader that comes with Arduino and just added my VID and PID; not an entirely new one

matthijskooijman, 2018-01-25:
Even so: This is not a problem with Arduino-supplied hardware and software, so this is not something Arduino can support. There might be other that can help on the forums, as suggested.

newpie18, 2018-01-25:
which Forums, do you mean Arduino Fromus ?

matthijskooijman, 2018-01-25:
Yup. avr-freaks might also be a good source, perhaps, or something on stackexchange (on the topic of windows drivers)?

newpie18, 2018-01-25:
okay I see, thanks