arduino/Arduino #7158: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude"

drinkypoo, 2018-01-25:
I'm trying to compile the sketch for https://github.com/Imposter/brite-arduino and it requires constexpr so I am trying to use the Arduino beta. Unfortunately, it does not seem to be able to find its own tools.

java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:796)
	at processing.app.SketchController.exportApplet(SketchController.java:767)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2254)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

facchinm, 2018-01-26:
Possibly a duplicate of https://github.com/arduino/Arduino/issues/7082 .
Would you mind adding the information about your system (OS, other versions of the IDE installed alongside, version of AVR core reported by Library Manager) there? Thanks!

closed 2018-01-26 by facchinm

fnguyen99, 2018-05-14:
I have similar error when attempting to upload to the Adruino. Already spent 4 hours researching and trying to fix this error to no avail.
Can someone please provide useful suggestion?

Fred

Arduino: 1.8.6 Hourly Build 2018/05/10 01:33 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 2492 bytes (7%) of program storage space. Maximum is 32256 bytes.
Global variables use 208 bytes (10%) of dynamic memory, leaving 1840 bytes for local variables. Maximum is 2048 bytes.

java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2197)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.