arduino/Arduino #7163: Installing Arduino on Linux - guide errors

linoskoczek, 2018-01-28:
Hello,

Short remarks concerning this instructions: https://www.arduino.cc/en/guide/linux

- **Add additional command to _Please Read..._ instructions**
To make Arduino work under Linux in some cases you have to use these 2 commands:
```
sudo usermod -a -G dialout <username>
sudo chmod a+rw /dev/ttyACM0
```
In the mentioned instructions, second line is missing. I believe it's worth adding.
_Reference:_ https://arduino-er.blogspot.com.tr/2014/08/arduino-ide-error-avrdude-seropen-cant.html

- **Change title**
from: Install the Arduino Software (IDE) on on Linux
to: Install the Arduino Software (IDE) on Linux

matthijskooijman, 2018-01-28:
> sudo chmod a+rw /dev/ttyACM0

This is really *not* a good idea:
 - It completely voids all permission checks on the serial device, giving everyone read and write checks. If you do this, there is no need to fiddle with groups at all.
 - This chmod only stays in effect until the device is unplugged, since when it is replugged, the device file is recreated with default permissions. This will confuse people, who will then resort to even worse plans, like running Arduino as root.

Doing the chmod can be useful in some cases *for debugging*, to doublecheck that there is indeed a permission problem, but it is never a meaningful solution.

linoskoczek, 2018-01-28:
That's true, it's unsafe workaround, but atm I don't see any other way to get Arduino to work without IDE which throws this error:
_avrdude: ser_open(): can't open device "/dev/ttyACM0": Permission denied
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions._

bengtmartensson, 2018-01-28:
The correct way to solve this on a modern Linux is (AFAIK) to use udev, and add a udev configuration file that tells the udev system how to handle USB-Arduinos (identified by their USB attributes). I use the following file as `/etc/udev/rules.d/10-arduino.rules`

    SUBSYSTEMS=="usb", ATTRS{idProduct}=="0043", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_uno arduino_uno_$attr{serial}"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="7523", ATTRS{idVendor}=="1a86", SYMLINK+="arduino arduino_nano_qinheng"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="2303", ATTRS{idVendor}=="067b", SYMLINK+="arduino arduino_nano_prolific"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="6001", ATTRS{idVendor}=="0403", SYMLINK+="arduino arduino_nano_ftdi arduino_nano_ftdi_$attr{serial}"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="8037", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_micro"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="8036", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_leonardo"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="0042", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_mega arduino_mega_$attr{serial}"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="804f", ATTRS{idVendor}=="2a03", SYMLINK+="arduino.org arduino_m0_pro"
    SUBSYSTEMS=="usb", ATTRS{idProduct}=="2111", ATTRS{idVendor}=="03eb", SYMLINK+="arduino_m0_pro_prog"

which can distinguish between (most) different Arduinos. (Although the device seen by the Arduino IDE still is /dev/ttyUSB* or /dev/ACM*.)

There "must" be am udev tutorial out there that can be either "stolen" or linked to...

q2dg, 2018-02-11:
Why not provide an official udev's rule file?

kengdahl, 2018-03-12:
I'm going to close this and y'all will have to pull the beta branch or wait for the releas

closed 2018-03-12 by kengdahl

sergiobenrocha2, 2019-03-24:
Hello! Using Ubuntu 18.04 here, I needed to include TAG+="uaccess" to get it working:

```
SUBSYSTEMS=="usb", ATTRS{idProduct}=="0043", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_uno arduino_uno_$attr{serial}", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="7523", ATTRS{idVendor}=="1a86", SYMLINK+="arduino arduino_nano_qinheng", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="2303", ATTRS{idVendor}=="067b", SYMLINK+="arduino arduino_nano_prolific", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="6001", ATTRS{idVendor}=="0403", SYMLINK+="arduino arduino_nano_ftdi arduino_nano_ftdi_$attr{serial}", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="8037", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_micro", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="8036", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_leonardo", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="0042", ATTRS{idVendor}=="2341", SYMLINK+="arduino arduino_mega arduino_mega_$attr{serial}", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="804f", ATTRS{idVendor}=="2a03", SYMLINK+="arduino.org arduino_m0_pro", TAG+="uaccess"
SUBSYSTEMS=="usb", ATTRS{idProduct}=="2111", ATTRS{idVendor}=="03eb", SYMLINK+="arduino_m0_pro_prog", TAG+="uaccess"
```