arduino/Arduino #7164: New BlueDot Library for Arduino - BME280+TSL2591

BlueDot-Arduino, 2018-01-28:
Could you please add the BlueDot_BME280_TSL2591 library to the Library Manager?
https://github.com/BlueDot-Arduino/BlueDot_BME280_TSL2591

Thanks!

per1234, 2018-01-28:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

BlueDot-Arduino, 2018-01-28:
Sorry, forgot to create the release. 
Just created the public release version.

cmaglie, 2018-01-31:
The name `BlueDot BME280+TSL2591 Library` is not allowed because it contains the `+` character.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Please change the name written in the `library.properties` to `BlueDot BME280 TSL2591` (no need to add `Library` there, it's implicit), commit & push, and redo the release.

BlueDot-Arduino, 2018-01-31:
I changed the library name, created a new release and updated the README file. I hope now is everything right...

per1234, 2018-01-31:
You didn't update the `version` value in your library.properties file before creating the release. So the release version is V1.0.1 but the `version` value is set to 1.0.0:
https://github.com/BlueDot-Arduino/BlueDot_BME280_TSL2591/blob/V1.0.1/library.properties#L2

BlueDot-Arduino, 2018-02-01:
Hi! okay, I created a new release version (V1.0.3) and now it matches the version value. Did I miss anything? :)

cmaglie, 2018-02-05:
Done!

closed 2018-02-05 by cmaglie