arduino/Arduino #7168: [Library Manager] ArduinoSensors library

YuriiSalimov, 2018-01-29:
Could you please add the ArduinoSensors library to the Library Manager?
https://github.com/YuriiSalimov/ArduinoSensors

Thanks!

cmaglie, 2018-01-31:
There is already another library called `ArduinoSensor`.

Please chose another name possibly without the using the word `Arduino` (since you're using the Arduino IDE it is implicit that the library is for Arduino) and the word `Sensor` alone (because it's too generic and broad).

YuriiSalimov, 2018-01-31:
Ok, I will format my lib, divide it into smaller libraries. And in future will add new issue. Thanks.

closed 2018-01-31 by per1234

cmaglie, 2018-01-31:
Please consider adding a prefix to each library, so your toolkit can be easily identified, for example:
`YS Relay`
`YS Thermistor`
etc. etc.