arduino/Arduino #7170: A completely outdated Troubleshooting page

per1234, 2018-01-29:
https://www.arduino.cc/en/Main/Troubleshooting

Every bit of information on this page is completely outdated so it should be deleted. This is the 3rd Google search result I get for a search of `arduino troubleshooting`. 

Note this is a different page from https://www.arduino.cc/en/Guide/Troubleshooting with completely different content. Maybe https://www.arduino.cc/en/Main/Troubleshooting could be made to redirect to https://www.arduino.cc/en/Guide/Troubleshooting or just replace the content on Maybe https://www.arduino.cc/en/Main/Troubleshooting with a link to https://www.arduino.cc/en/Guide/Troubleshooting if you don't want to remove that page altogether.

kengdahl, 2018-03-01:
I replaced the content of: https://www.arduino.cc/en/Main/Troubleshooting
With the content of: https://www.arduino.cc/en/Guide/Troubleshooting

per1234, 2019-01-27:
https://www.arduino.cc/en/Main/Troubleshooting still contains the same completely outdated content. I don't know whether the update never went through or if the change ended up being reverted later. It would be best if the page was redirected to https://www.arduino.cc/en/Guide/Troubleshooting.

animeshsrivastava24, 2019-06-03:
@per1234  Is this issue assigned to someone? I'll be happy to document it and get reviewed.

per1234, 2019-06-04:
@animeshsrivastava24 the removal of the obsolete page and redirect of that URL to the more modern troubleshooting page can only be done by someone with edit access to arduino.cc so there is no way you can assist with this issue.

animeshsrivastava24, 2019-06-04:
@per1234 Thank you so much for the reply. Sure, I can understand.

kengdahl, 2019-06-26:
@animeshsrivastava24 I updated the page again. It should be good now.

closed 2019-06-26 by kengdahl