arduino/Arduino #7171: Add new Temperature_LM75_Derived library

jeremycole, 2018-01-30:
Please add the following new library:

https://github.com/jeremycole/Temperature_LM75_Derived

cmaglie, 2018-01-31:
Hi @jeremycole,

the name `I²C Temperature Sensors derived from the LM75` contains the UTF-8 char `²`, even if not explicitly denied it may cause some troubles, would you like to replace the `²` with a `2`?

jeremycole, 2018-01-31:
Hi @cmaglie! Not a problem at all. Replaced I²C with I2C in the name and bumped the version.

jeremycole, 2018-02-04:
FYI, we have now tagged version 1.0.0 of this library, after receiving, building breakout boards for, and testing it with 14 additional LM75-derived temperature sensors!

[image]

cmaglie, 2018-02-05:
Done!

closed 2018-02-05 by cmaglie