arduino/Arduino #7174: new lib for hcsr04 sensor

gamgine, 2018-01-30:
Please add the hcsr04sensor library to the library manager list. Thanks.
https://github.com/gamgine/Arduino-lib-HCSR04/releases
https://github.com/gamgine/Arduino-lib-HCSR04

cmaglie, 2018-01-31:
There is already another library called `HCSR04`, please choose another name.
Remeber to update the `library.properties` to reflect the new name and re-do the release tag **after** the update to `library.properties` is commited and pushed.

gamgine, 2018-02-01:
I'm coming to find nothing of that name in the libraries manager.
but name "ultrasonic sensor" is available?

cmaglie, 2018-02-05:
`HCSR04` is taken by https://github.com/Martinsos/arduino-lib-hc-sr04.git but for some reason is not being published, will look at this separately.

About your library, `Ultrasonic Sensor` is available, but I will use `HCSR04` in the name, maybe prefixed with something else like `gamgine_HCSR04`... anyway, up to you.

gamgine, 2018-03-02:
hi sorry for long time
I chose to use `hcsr04 ultrasonic sensor` if is available for name

cmaglie, 2018-03-03:
Ok, please:

- update name and version in your `library.properties`
- commit & push
- make a new release / git tag

gamgine, 2018-03-04:
ok it's done
library.properties , release (1.0.0) and readme
https://github.com/gamgine/HCSR04-ultrasonic-sensor-lib

cmaglie, 2018-03-08:
Done!

closed 2018-03-08 by cmaglie