arduino/Arduino #7177: Proposal: Could String::toFloat() return NAN instead of 0.0 on fail?

ScruffR, 2018-01-31:
I'm aware that this might cause some trouble changing and a similar request once before was turned down without real reason (https://github.com/arduino/Arduino/issues/1796), but 0.0 may well be a desired result making it difficult to distinguish between a failure from the actual value. Since for that case extra tests are required one might just stick with the extra test and not at all check for 0.0.

However if breaking existing code has to be avoided at all cost, how about adding an optional bool* success or even double* retValOrNAN?

Same for String::toInt(), Stream::parseFloat() and Stream::parseInt().

ScruffR, 2018-01-31:
I guess the `Type: Duplicate` refers to the already (and still) closed issue #1796 without considering the extra part about the *optional parameter*, right?

per1234, 2018-01-31:
The topics of both issues are the same. Yes, you have suggested some new options here but you also commented about that in https://github.com/arduino/Arduino/issues/1796 so there is no unique content in this issue. However, https://github.com/arduino/Arduino/issues/1796 was closed without giving a clear reason and that was done prior to your added suggestion so I won't close this for now in case it might prompt one of the Arduino team to take another look at https://github.com/arduino/Arduino/issues/1796 and consider either reopening it or commenting on why it should remain closed.

ScruffR, 2018-01-31:
@per1234, fair enough, I was just concerned that the duplicate tag would prevent others from taking notice. If it helps I could just remove the comment on the old issue (where I posted first before deciding to file it as an issue in its own right), since it won't do any good there now anyway.

per1234, 2018-01-31:
Leave things as is for now and let's see what happens in the coming days. It's best to keep everything in a single issue, otherwise the discussion is fragmented and important information may be lost in all the thousands of issue reports. https://github.com/arduino/Arduino/issues/1796 contains more information than this one so that issue is preferable to this one.

ayohenry, 2018-02-25:
Where can I find script to attempt to fix this issue in the repo?

closed 2018-02-28 by per1234

kubark42, 2018-04-07:
@per1234 Agree with everything you say, but I notice that you closed this issue without opening #1796. Think you could re-open there, or leave a comment as to why it's a WONTFIX?