arduino/Arduino #7195: Changed Arduino library name. Please update.

jaean123, 2018-02-06:
I made this request to change my library name and made the fixes to the ".properties" file.
It's been a couple of days, but I still could not find the library in the library manager.
Could someone get this fixed soon.

Thank you.

https://github.com/arduino/Arduino/issues/7162

cmaglie, 2018-02-06:
@jaean123 
maybe I didn't explain it clearly enough: you changed the name in `master` branch but NOT in the github release `1.0` (that is where it counts), see here: https://github.com/jaean123/ArduinoBlue-library/blob/1.0/library.properties#L1

To fix this, as I already explained in my [latest post](https://github.com/arduino/Arduino/issues/7162#issuecomment-361923789) you must remove the git tag/release `1.0` and re-do it on `master`, but **you** are the only one that can do it. Once done the library will be automatically published (within an hour).

jaean123, 2018-02-06:
Sorry about this. I thought I changed it on 1.0 as well since I pushed to the "release" branch.
I see it on the library manager. Thanks!

closed 2018-02-06 by jaean123