arduino/Arduino #7200: [Library Manager] register request: DynamixelShield library

OpusK, 2018-02-09:
Hi. we made Arduino library for [DynamixelShield](http://en.robotis.com/model/page.php?co_id=actuator).
DynamixelShield is an actuator(specific to Dynamixel) shield and I think this library helps using your Arduino board with DynamixelShield easily and fast.
Please add our library to Library manager.

```
This is github url : https://github.com/ROBOTIS-GIT/DynamixelShield
```

Thank you :)

per1234, 2018-02-09:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

OpusK, 2018-02-09:
Oh, I forgot.
As your request, i created [tag](https://github.com/ROBOTIS-GIT/DynamixelShield/releases/tag/0.0.1).

Thank you :)

OpusK, 2018-02-09:
Your pull request was not included in our 0.0.1 release, so we released it in 0.0.2.
Thanks!

cmaglie, 2018-02-13:
Done!

closed 2018-02-13 by cmaglie