arduino/Arduino #7203: Rename tab results in link error

sterretje, 2018-02-10:
I could not find this one; also not sure if it belongs in this section but I saw a few more relating to tabs so here it goes.

1. Compile a sketch with multiple tabs.
2. Rename one of the cpp tabs (I renamed both a cpp and h file and adjusted the main sketch accordingly).
3. Compile again.

The result is an error about multiple definition of the functions where the original object file is mentioned.

Current workaround is to close and open the IDE (using version 1.8.5)

per1234, 2018-02-10:
Please test with the beta build of the Arduino IDE (https://www.arduino.cc/en/Main/Software#beta) and let us know whether it still occurs. I believe this issue was fixed after the 1.8.5 release.

sterretje, 2018-02-10:
That will have to wait for somewhere next week; I want to update a machine anyway so that will be a good candidate to test it on.

sterretje, 2018-02-16:
OK, that seems to be solved.

Where do I report bugs on the beta release?

closed 2018-02-16 by sterretje

per1234, 2018-02-16:
@sterretje any testing of the beta build is very welcome! You can report any bugs you might find in the same places as always. The appropriate repository depends on the topic.

I've attempted to put together a guide to the various places you can submit issue reports here:
https://github.com/arduino/Arduino/blob/e4509d3daf41ba4a29b2867f64828dd129e61965/CONTRIBUTING.md#issues

If you have any suggestions for improvements to that document I would welcome them here: https://github.com/arduino/Arduino/pull/6357.

Since the time I wrote that list there is a brand new tool in the beta build named arduino-preprocessor. This does all the sketch [preprocessing](https://github.com/arduino/arduino-preprocessor#about-the-arduino-sketch-ino-preprocessing), which was previously handled by arduino-builder (or just the IDE in Arduino IDE 1.6.5 and older). So if you find a bug with, for example, function prototype generation the place to report it is here:
https://github.com/arduino/arduino-preprocessor